ESPresense/ESPresense #142: Allow manual entry of WiFi credentials from webserver (ESP32 configuration)

yang3535, 2021-12-05:
**Is your feature request related to a problem? Please describe.**
Related to issue #141, but should done for hidden WiFi networks regardless

**Describe the solution you'd like**
Allow manual entry of WiFi credentials from webserver page (192.168.4.1)

**Describe alternatives you've considered**
Add option to add WiFi credentials manually independently of scanning for networks.  

**Additional context**
Add any other context or screenshots about the feature request here.

FizzedInMyPantz, 2022-01-06:
Agreed, this would be helpful when connecting to networks with a hidden SSID.

corgan2222, 2022-01-09:
Fixed in my fork
https://github.com/corgan2222/ESPresense

closed 2022-04-10 by stale[bot]