ESPresense/ESPresense #151: Detecting Apple Watch6-2

vogon1, 2021-12-17:
I got myself a number of Atom Lite ESP32 devices, and within minutes they were up and running.
My main goal is to track Apple Watches. That works quite well for detecting Apple Watch series 4, but not so well for series 6.

The Apple Watch 6 is detected quickly as apple:1005:9-26.
Some minutes later (10 to 20 minutes), it shows up as apple:watch6-2.
Then, after MAC address rotation, it is back to apple:1005:9-26, and again after some minutes it is apple:watch6-2.

Is there a  way get the detection of apple:watch6-2 quicker?

Sietse

mguellsegarra, 2021-12-22:
First at all, thanks to all the contributors for this magnific project.

@vogon1 I've got this weird behaviour too, not sure if I miss something. I've got an Apple Watch Series 7, which is reported as `apple:watch6-6`, `apple:1005:9-26`, `apple:1005:9-12`, ... And I'm not sure which one I have to use, I got erratic presence reports (`not_home` while being at 1m of my ESP32) no matter which one I pick.

I'd be happy to understand the magic that this project do internally but sadly I have not the knowledge... yet!

Let us know if we can help someway with this.

Thanks!

nmajin, 2022-01-17:
I also see an issue like this as well at times where it appears my watch (Series 7) falls off the monitoring and doesn't report it's state (what room it is). I noticed this option off by default in the ESPresense config page (http://<IP>):
- "Query devices for characteristics (helps apple fingerprints uniqueness) (default: ☐)"

Maybe it can help but I haven't played around with it enough.

DTTerastar, 2022-01-17:
I think this is because once a ton of nodes slam the watch w/ queries to identify itself, the watch thinks it's getting DDOSed and refuses to answer.  The best solution would be for the ESPs to identify a node as closest and just have that node query and then send the answer to every node.  That requires the code to get a lot more clever... Thus why I haven't gotten it done yet...

closed 2022-04-10 by stale[bot]