ESPresense/ESPresense #154: Implement Home Assistant Autodiscovery for mqqt_room

bluesheep, 2021-12-22:
**Is your feature request related to a problem? Please describe.**
I would like devices that I'm tracking to be automatically discovered and configured within home assistant.

**Describe the solution you'd like**
Newly discovered BLE devices should be able to publish their MQTT information in a format that allows home assistant to automatically discover them as new devices. 

**Describe alternatives you've considered**
The existing process of manually editing sensors is functional but suboptimal.

**Additional context**
In high traffic areas this is likely to result in a lot of devices being discovered within home assistant. Consideration should be made for whether a whitelist/blacklist solution is required.

corgan2222, 2021-12-22:
I will look into this.

bluesheep, 2021-12-22:
@corgan2222 let me know if I can do anything to help. I've got the code compiling and uploading locally. I'm OK at writing code but I have no experience with GitHub, and only hobby hacking with c++.

corgan2222, 2021-12-22:
Thanks. ATM im working on support for more input/output pins, like LED, switches, aso.
After that, I will look into this.

DTTerastar, 2022-08-30:
With v3 we now have the beginning of a ui (see /ui/) that I intend to use to make this happen.  Stay tuned!

bluesheep, 2022-08-30:
Woohoo!

DTTerastar, 2022-09-01:
next!

bluesheep, 2022-11-01:
Still keen. Only replying to distract the stalebot

closed 2023-03-25 by stale[bot]