ESPresense/ESPresense #156: To be able to disable the onboard LEDs

circa1741, 2021-12-27:
**Is your feature request related to a problem? Please describe.**
Some onboard LEDs are too bright especially in the bedroom during sleeping time.

**Describe the solution you'd like**
It would be nice to be able to disable the onboard LEDs. For boards with multiple LEDs, maybe a comma-separated GPIO values to disable.

**Describe alternatives you've considered**
Black nail polish, black electrical tape, removing the LEDs, etc.

**Additional context**
N/A

FizzedInMyPantz, 2022-01-06:
Agreed. I've resorted to covering with a sock lol would love to be able to turn off LEDs via software.

mjcumming, 2022-01-11:
Please!

nmajin, 2022-01-17:
I agree, luckily the way I have it plugged in the LEDs are facing the wall but they are still lighting up the rooms.

DTTerastar, 2022-01-17:
I can't control the power leds on the devkits.  So no matter what you've got one led to cover up :(

commit 3a54825c (2022-01-23): Allow status led to be disabled (fixes #156) (#196)
closed 2022-01-23 by DTTerastar