ESPresense/ESPresense #159: QryErr on Apple Watch

linglingpin, 2022-01-01:
I installed ESPresense v2.0.13 on a M5 Atom Lite yesterday, every thing works at first. And now somehow it's having trouble to query an Apple Watch, and it makes detecting other devices (also with close/left) having trouble. Switching off query takes the error off the log, but some devices (ie iPhone) become unresponsive after a while. Any help is greatly appreciated!!

> BH1750_I2c Sensor: 
Connecting to MQTT 10.15.20.30 1883
1 Model | MAC: 7a4f6e3ff450, ID: apple:ipad7-1                                                iPad7,1
1 Model | MAC: 7efb037209ac, ID: apple:ipad13-4                                               iPad13,4
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 0, retry after 1ms
E NimBLEClient: "Client busy, connected to 6f:1e:68:1e:7d:3e, id=0"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 1, retry after 2ms
E NimBLEClient: "Client busy, connected to 6f:1e:68:1e:7d:3e, id=0"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 2, retry after 4ms
E NimBLEClient: "Client busy, connected to 6f:1e:68:1e:7d:3e, id=0"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 3, retry after 8ms
E NimBLEClient: "Client busy, connected to 6f:1e:68:1e:7d:3e, id=0"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 4, retry after 16ms
E NimBLEClient: "Client busy, connected to 6f:1e:68:1e:7d:3e, id=0"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 5, retry after 32ms
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 6, retry after 64ms
E NimBLEClient: "Client busy, connected to 6f:1e:68:1e:7d:3e, id=0"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 7, retry after 128ms
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 8, retry after 256ms
E NimBLEClient: "Client busy, connected to 6f:1e:68:1e:7d:3e, id=0"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 9, retry after 512ms
0 New   | MAC: 3b62b47dee66, ID: md:00e0:8                                                    
0 New   | MAC: 7a4f6e3ff450, ID: apple:ipad7-1                                                0d1dfa605f78
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 10, retry after 1024ms
E NimBLEClient: "Client busy, connected to 6f:1e:68:1e:7d:3e, id=0"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 11, retry after 2048ms
0 New   | MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              0d18f93450
0 New   | MAC: 7efb037209ac, ID: apple:ipad13-4                                               011dd46d7e68
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -52, try 12, retry after 4096ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -56, try 13, retry after 8192ms
0 Close | MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              (0.52m) -32dBm
0 Left  | MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              (0.51m) -53dBm
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -53, try 14, retry after 16384ms
0 New   | MAC: 55a7ce34a358, ID: apple:1006:10-12                                             0e1d9c40fd18
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: 55a7ce34a358, ID: apple:1006:10-12                                             rssi -88, try 0, retry after 1ms
1 Model | MAC: 55a7ce34a358, ID: apple:ipad11-2                                               iPad11,2
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -49, try 15, retry after 32768ms
0 New   | MAC: c869cd5ece8b, ID: apple:1005:9-12                                              01143db606
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -59, try 16, retry after 65536ms
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -50, try 17, retry after 60000ms
1 Model | MAC: 5b11f38da828, ID: apple:ipad13-4                                               iPad13,4
0 New   | MAC: 5b11f38da828, ID: apple:ipad13-4                                               011d8cd8b318
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 6f1e681e7d3e, ID: apple:1005:9-24                                              rssi -57, try 18, retry after 60000ms
Not updating from (sc=302): https://github.com/ESPresense/ESPresense/releases/download/v2.0.13/esp32.bin

remi-gelinas, 2022-01-07:
I'm having this issue now too with all Apple devices connected to my ESP32. Same version. iPhone, Apple Watch, iPad, and Macbook all produce this exact error:

```
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
```

linglingpin, 2022-01-07:
To give more context, I too later figured many of my Apple devices exhibit this problem too instead of just the Apple Watch. The detection is very slow / unreliable even with query off (although the errors reported above has disappeared).

I tested by scanning with Bluetility last night and find there are no discovery problems using my Mac. However, I find there are like 30 bluetooth devices nearby as I live in Hong Kong where the living spaces are very congested, would the total number of devices nearby affect the scanning performance of the ESP devices? In addition, I have also set up another ESPresense node with an M5Stack CPlus since, but the same issues occurs.

I have also set up 2.4 GHz band only for my IoT Wi-Fi network but it didn't help with the discovery problem, however the uptime seems to improve.

Furthermore, I also set up room-assistant addon with Home assistant installed on an RPi 3 B+, the discovery seems much faster and much more reliable. But I really wish to use ESPresense in the long run as I don't have that many RPis laying around.

I will try manually installing older versions of ESPresense to check when I have the time.

DTTerastar, 2022-01-17:
Yeah, trying to query every single apple device in a location that congested is going to lead to issues.  We really need a central brain to know what to bother querying.  I'm thinking of trying more of a [Passive BLE Monitor](https://github.com/custom-components/ble_monitor) approach where we make the nodes dumber, and a controller smarter.

closed 2022-04-10 by stale[bot]