ESPresense/ESPresense #163: Support for ESP32 cam

craigjmidwinter, 2022-01-03:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Support for the ESP32 Camera module in order to avoid needing to deploy multiple ESP32 units in a single room

**Describe alternatives you've considered**
Deploying multiple ESP32

**Additional context**
[Example of ESP32 cam project](https://github.com/easytarget/esp32-cam-webserver)

DTTerastar, 2022-01-17:
There is an attempt at this in a PR, but the poor CPU of the ESP starts to get bogged down when trying to also support images.

duceduc, 2022-01-22:
Would it be ok to use the ESP32 cam strictly for BLE tracking? Meaning flashing the esp32 cam with espresense firmware.

closed 2022-04-10 by stale[bot]