ESPresense/ESPresense #172: Connect to hidden network/static ip

marrobHD, 2022-01-10:
**Is your feature request related to a problem? Please describe.**
I have my IoT WiFi hidden, and there's as far as I know no easy way to connect to it manually and use a static ip. 

**Describe the solution you'd like**
Ability to manually type SSID in captive portal and ability to manually turn on static ip.

**Describe alternatives you've considered**
/

**Additional context**
It could be easier to implement this with the [ESP Async Web Server library](https://github.com/me-no-dev/ESPAsyncWebServer). The same library is used at Tasmota.

siiiilky, 2022-01-11:
I've compiled this from source and have an option to connect to a hidden (manual) SSID. I've not tried it as I don't use hidden SSIDs as they don't achieve anything.

Any reason why you don't just use DHCP reservations instead of a static IP?

marrobHD, 2022-01-11:
I hide the ssids because every iot has a different ssid

I prefer using a static IP for performance reasons.

siiiilky, 2022-01-11:
How many SSIDs do you advertise? In general, it's recommended not to have more than 3.

Performance reasons for a static IP?

marrobHD, 2022-01-11:
> How many SSIDs do you advertise? In general, it's recommended not to have more than 3.
> 
> Performance reasons for a static IP?

3 SSIDs per Access Point. Two are hidden. 

On some esp devices/WiFi libraries the WiFi connection is unstable at hidden networks when not defining a static IP.

DTTerastar, 2022-01-17:
Sorry but this need is a bit too niche.  But you've got the source so you can go nuts and hardcode a bunch of stuff!

closed 2022-01-17 by DTTerastar