ESPresense/ESPresense #173: Can I name two base stations the same to cover a longer room

JoeGandy, 2022-01-10:
I have a room that no matter where i put the sensor, it lose connection on one end, If I put two base stations in the room at each end, and name them the same does that work?

Or should I just have like lounge_1 and lounge_2 and handle it that way?

Maybe some updates to the docs could be made based on the solution to this

DTTerastar, 2022-01-17:
What kind of device are you tracking?  Can you instead increase the power output, or increase the max distance?  I usually find I have the opposite problem of lots of rooms seeing the same device.  Or are you living in a mansion?

closed 2022-04-10 by stale[bot]

gith-account, 2022-11-28:
See issue #600.

[DTTerastar](https://github.com/DTTerastar) (Maintainer) [on Aug 22](https://github.com/ESPresense/ESPresense/discussions/600#discussioncomment-3450043):
> Room names are required to be unique because we use it for our mqtt client id.