ESPresense/ESPresense #174: Not a bug... Just a newbie question

hvddrift, 2022-01-11:
I have some of these: ATOM Lite from M5 

Which install selection do I pick?

The M5Atom-Matrix-Led Matrix? 

[image]
g

Thanks

mjcumming, 2022-01-11:
This only works on an ESP32

hvddrift, 2022-01-11:
This is a ESP32 device. It's the top one listed on this page

https://espresense.com/hardware

Arganox, 2022-01-13:
@hvddrift have you tried flashing it with the generic ESP32 image? I'm using this on a M5Stamp Pico.

StarFleetCPTN, 2022-01-13:
I used m5atom-matrix on my M5Atom lites without any changes.

begunfx, 2022-01-14:
I have the Atom lite and I just used the generic esp32.

hvddrift, 2022-01-17:
Thank you. Works great.

Arganox, 2022-01-17:
> Thank you. Works great.

Good to hear! 👍 
Don't forget to mark the issue as solved.

closed 2022-02-13 by hvddrift

DTTerastar, 2022-02-21:
Yes, The M5Atom-Matrix-Led works great on all m5atoms.