ESPresense/ESPresense #177: Feature Request - Apple AirTags

LordNex, 2022-01-12:
### Discussed in https://github.com/ESPresense/ESPresense/discussions/150

<div type='discussions-op-text'>

<sup>Originally posted by **LordNex** December 15, 2021</sup>
I was wondering if there was going to be support for Apple AirTags as well? We use them here as we are all in an Apple eco system with our mobile devices anyway so it makes tracking things like my keys and wallet real easy and shareable with family in FindMy. I know they kind of integrate through the iCloud HA Integration but i havent been able to track them as far as location</div>

DTTerastar, 2022-01-17:
I don't think this is possible using just BTLE broadcasts due to all of the anonymity that Apple has baked into how they work.

LordNex, 2022-01-18:
I know they show up as devices if I install the  iCloud integration. But it doesn’t really give the ability to track them. But I figured if there was a way to get them in through that integration, you may be able to work with that for authentication. Just a thought anyway. I love mine as I get way more use out of them than I do with the other beacons I’ve tried. And a 4 pack for $99 is pretty good considering that they are  and have a user replaceable battery.

Benjamin T Powers

closed 2022-04-10 by stale[bot]

LordNex, 2022-04-11:
So does that mean it can’t be done?

DTTerastar, 2022-04-11:
I'm sure some security researcher will eventually crack it.  But hasn't happened yet AFAIK.