ESPresense/ESPresense #179: My evaluation of Android/IPhone as beacons

begunfx, 2022-01-14:
I have one IPhone and two Android phones in the house.  So far I've setup 5 of my 10 sensors and have been playing with how to connect our three smart phones.  The IPhone seems to show up without much effort and seems to be easily identified.  Our Android phones however have been very problematic.  I've tried BeaconScope and BeaconSimulator.  On my first try at trying to use an IBeacon from one of the apps, it showed up on my list, but only 1x.  I haven't been able to see an IBeacon since.  I have had success using Eddystone-UID (It appears as "sd:").  The biggest issue with using Eddystone-UID is that if I have it on both Android phones, they seem to both list as "sd:" with no differentiation.   I have also been able to use AltBeacon to show up (It appears as "md:0118:26").   But the issue with both of these is I can only use them on one device.  If I use the same type of beacon on both devices, they appear the same on the list, just in different rooms.  The only solution I could come up with is one of the phones uses Eddystone and the other AltBeacon.  But that would be the extent of expandability.  If at some point I want to add another android phone, then it seems I'm out of luck.  Does anyone have better luck with multiple Android phones?

DTTerastar, 2022-01-17:
I added Eddystone-UID and AltBeacon support just now so this should be fixed.  Also, have you tried the Home Assistant Companion App (on Android)?

begunfx, 2022-01-17:
@DTTerastar awesome.  Thanks!  I'll play with it once the ESP32s update.  Unfortunately, I don't have a Home Assistant System.  I'm running HomeSeer.  I'm able to grab the data from the ESP32s using MQTT plugins.  I'm still trying to figure out the detection logic since I don't have access to Home Assistant.

begunfx, 2022-01-17:
@DTTerastar Not sure what changes you made to the latest release, but I cannot see Eddystone-UID or AltBeacons at all anymore.  in 2.0.13 I was able to see Eddystone-UID and AltBeacons.  I just couldn't differentiate them in multiple devices because the payload didn't contain the id tags.   When you updated to 2.0.14, I no longer can see Eddystone-UID or AltBeacons at all now.  I'm using MQTT explorer on my Mac to watch the MQTT traffic on my HomeSeer MQTT plugin/broker.

DTTerastar, 2022-01-17:
[image]
[image]

Set your maxDistance to 0 (meaning to not filter anything out) just to make sure that's not why you're not seeing it.

begunfx, 2022-01-17:
I tried filtering MaxDistance to zero but no luck.  For some strange reason, before you added official support for EddyStone-UID I could see it on my list as "sd:" but now I don't see that anymore and do not see "eddy".  As you can see from my screengrab.  All of those devices can be accounted for
[image]
.

DTTerastar, 2022-01-17:
md:0x00E0 is Google.  Is that your phone?  If you hook the serial console up and bring the phone close to the esp you can figure out which is which.  8 bytes isn't enough to have a beacon on it.

md:ffff:26 is what an undecoded altBeacon should look like.  But :3 means we're only seeing 3 bytes instead of 26.  Very odd.

Another thing to try is the verbose firmware which will log all of the bytes of all the advertisments seen.

begunfx, 2022-01-17:
the md:ffff:3 is my Ember Mug and the md:0x00E0 devices are most likely my Google Chromecasts which I have in several rooms of the house.
[image]

See the attached detail of the signal.

begunfx, 2022-01-17:
Okay.  I'll try flashing my office sensor using verbose since that's where I'm spending most of my time etc.

begunfx, 2022-01-18:
I've noticed the following message a few times in the console when re-flashing my esp32 sensors

lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "

DTTerastar, 2022-01-18:
Those are basically saying we couldn't query a device. The ESP retries so it's not a big deal.

begunfx, 2022-01-18:
Okay.  Good to know.  Just thought it was worth sharing.

begunfx, 2022-01-18:
Okay.  So I reflashed all my sensors.  So, still not out of the woods.  Still not sure what's going on.  I can't see any of the beacons again in the MQTT broadcast.  I can see them when I have the sensor (any of the sensors) hooked up to my Mac and running the console.  But when I have the plugged into their locations, I'm not capturing any beacons.

begunfx, 2022-01-18:
Another update:  If I plug a sensor into my Mac then the beacon appears in MQTT, but if I plug it into a wall USB, nope.

begunfx, 2022-01-18:
See #182 that resolves my open concerns.

closed 2022-01-18 by begunfx