ESPresense/ESPresense #181: distinguish multiple (e.g.) iPhones/iPads/Apple Watches of same model?

wivaku, 2022-01-14:
Don't think it is a bug. But could not find this in the (troubleshooting) documentation, apologies if this is a beginner's question.

Using ESPresense v2.0.13 and e.g. two iPhones (both iPhone 12 Pro).

Both appear using topic `espresense/devices/apple:iphone13-3`.  
Is there a way to distinguish them and treat them as two beacons / topics?  
I can imagine this is tricky as these devices regularly change their address.

Same for two iPads / Apple Watches of the same model.  
In my case we have two different iPad models, which works fine.

In the ESPresense UI I have enabled all checkboxes, including:
- active scanning (uses more battery but more results)
- query device for characteristics (helps apple fingerprints uniqueness)

DTTerastar, 2022-01-14:
Install the room asssistant iOS App and your phone will get a unique guid.

wivaku, 2022-01-14:
Thanks, this indeed works!  
I now see: `espresense/devices/roomAssistant:8e521c5a-...-1ae9ae2a233c`

For Android I now see you can also use the [Home Assistant companion app](https://companion.home-assistant.io/docs/core/sensors/) and use the BLE Transmitter sensor for that purpose (but: not available for iOS).

Suggestions for the Apple Watch?

DTTerastar, 2022-01-14:
Apple Watch there isn't a solution for...

wivaku, 2022-01-14:
ok thanks

closed 2022-01-14 by wivaku

Inrego, 2022-01-17:
Does the Room Assistant iOS app work in the background now? Or is it still pretty useless in regards to actually tracking an iPhone?

DTTerastar, 2022-01-17:
yes, it works even tho it's backgrounded.  It doesn't launch automatically on reboot tho.

Inrego, 2022-01-21:
> yes, it works even tho it's backgrounded.  It doesn't launch automatically on reboot tho. 

When did it get fixed? I tried a few months ago, and it was very unreliable