ESPresense/ESPresense #182: Allow the ESP32 sensor to be paired with an Android phone to track the device

begunfx, 2022-01-14:
**Is your feature request related to a problem? Please describe.**
The current issue is the challenge of tracking Android devices.  Using beacon apps seem to really not help much or at all.

**Describe the solution you'd like**
How about the possibility of allowing the ESP32 sensors to pair with an Android device?   If the issue is not being able to see the Android device during scanning due to the randomization of the MAC address, then if the devices were paired, wouldn't the scanning issue go away?  Forgive me for my lack of knowledge of how BLE works, but the thought is that if the devices are paired then the ESP32 sensors will be able to track the Android device.

DTTerastar, 2022-01-17:
The[ HA Companion app ](https://play.google.com/store/apps/details?id=io.homeassistant.companion.android&hl=en_US&gl=US) can broadcast iBeacons in the background on android so we can track them.  AFAIK this makes tracking android phones a solved problem.

begunfx, 2022-01-17:
@DTTerastar unfortunately, I don't have a Home Assistant system.  I'm using HomeSeer.  In order to use iBeacons I need to use either Beacon Scope or Beacon Emulator which both broadcast proper iBeacons and for some reason they do not appear in MQTT messages.

begunfx, 2022-01-17:
@DTTerastar out of curiosity have you tried using the Beacon Scope or Beacon Simulator apps to detect iBeacon, Eddystone-UID or AltBeacons on Android? If not, can you please try?

DTTerastar, 2022-01-17:
I used Beacon Simulator to make AltBeacons and EddyStone-UID beacons

begunfx, 2022-01-18:
@DTTerastar okay.  So I flashed my office esp32 to be verbose and an interesting thing happened.  I'm seeing the iBeacon, Eddystone-UID and AltBeacon signals coming through.  I'm just not understanding why I don't see them in the non-verbose flashing.  See attached
[image]
[image]
[image]
.

begunfx, 2022-01-18:
Okay.  So after that test, I reflashed it back to non-verbose and it still shows up.  However, if I go to my other sensors, it's not picked up.  Something weird in the flashing?  How do you explain that?  I have 7 sensors all running.  I re-flashed my office to verbose and back and now all three beacons are reporting in verbose and non-verbose mode.  However, my other sensors that I haven't reflashed do not show the beacons.

DTTerastar, 2022-01-18:
Very very strange

begunfx, 2022-01-18:
As I had mentioned in the other thread.  It seems that the issue is when I plug the sensor into a wall outlet via USB, the Beacons are not reporting.  But if I plug in the sensor into my Mac USB port, they do.  What the hell!

begunfx, 2022-01-18:
I've tried different cables, and different power options (direct to USB in power outlet and a USB power adapter).  No luck.

begunfx, 2022-01-18:
One more update:  If I reset the beacon (using the reset button) the beacon appears in MQTT (but only 1x)

begunfx, 2022-01-18:
Okay.  So I played with it some more.  If I set my maxDistance back to the default 16 I seem to get the Beacons to appear.   I have a Wifi access point in the middle of my office (so only about 1.82 meters away from the sensor), and my desk and phone are only about .914 meters away from the sensor.  I had the setting on the sensor to about 3 meters.  So, I'm not sure what's going on.   Same deal with a couple of my other sensors.  I changed them to be the default 16 meters and as I walked by the beacon did appear on MQTT.

DTTerastar, 2022-01-18:
Beacon Simulator let's you set what the RSSI should be at 1m (iBeacon) or 0m (EddyStone).  I'd bet you put the 1m in the 0m beacon and thus why ESPresense thinks everything is super far away.  That's why at the beginning I set to set your maxDistance to zero so we could eliminate this class of problem :)  To switch between 1m to 0m add 41db.

To get accurate distance the best thing to do is set maxDistance to 0.  Put your beacon 1m away, look in the mqtt and put that value into beacon simulator (if an eddystone subtract 41).  You should then see the value in rss@1m be that value and the distance should read 0.

begunfx, 2022-01-18:
I was mostly using the Beacon Scope app (until you mentioned that you were using Beacon Simulator).  I tried just using the samples provided without changing anything so I'm not sure about the 1m/0m settings.  

I would strongly recommend that you put together a tutorial/wiki walking users through how to properly setup the sensors/beacons as I bet this is something a lot of people are running into.  I've seen others mention that they didn't see Android devices appear with the sensors, so it might be the same problem.

I'll try your diagnostic.  When you say: "put that value into beacon simulator"  Which value should I be putting in the beacon simulator?  What are you referring to?

Thanks!

begunfx, 2022-01-18:
Okay.  I think I understand what you are saying.  I set the maxDistance of my office sensor to 0 and put my Android phone 1m away.  I'm using an iBeacon in Beacon Emulator.  The iBeacon txPower is set to -65 (default).  Frequency mode is also at default of Balanced ~3hz.  What I'm not understanding is that the distance isn't reading as 1m.  It seems to be way off.  See below.

This is the resulting info:

{
"rssi@1m":-65,
"rssi":-99,
"raw":9.36,
"distance":8.88,
"speed":0.1
}

DTTerastar, 2022-01-18:
Set the iBeacon txPower (rssi@1m) to -99, then ESPresense will read the distance as 1m.  Basically the beacon allows for you to send out a calibration value so the receiver (ESPresense) can calculate your distance.  The confusing part is there is 2 values of txPower's in there, one which is how powerfully to broadcast (this is at 0meters), the other is the calibration value at 1m.

begunfx, 2022-01-18:
Okay.  Awesome.  That helps a lot.  I think I got that finally figured out.  Yeah, it is confusing.   Thanks for taking the time to help me through this.  Really appreciate it.  And again, awesome work putting this together.  Now the last thing  I need to figure out is how to get the mqtt_room logic in the HA plugin into HomeSeer.

DTTerastar, 2022-01-18:
Sweet Can you close a few of these issues now?

begunfx, 2022-01-18:
Will do.

begunfx, 2022-01-18:
Closing this thread as @DTTerastar helped identify some setup issues that prevented my Android Phone Beacons from being detected.

closed 2022-01-18 by begunfx