ESPresense/ESPresense #183: Add ids to Eddystone-UId and AltBeacon signals to identify unique Android Phones

begunfx, 2022-01-14:
**Is your feature request related to a problem? Please describe.**
Currently if you try to use an app like Beacon Scope for Android and use either Eddystone-UID or AltBeacon you get a generic reading that doesn't have any specific identifying parameters, so essentially you can only use each for one device. 

**Describe the solution you'd like**
From what I see in the data payload, they have ids.  Can you please try to include the ids in the broadcast?  I'm hoping that could be a way to differentiate multiple Android phones all running the same type of beacon.

Note: I'm not an expert or authority on the subject of BLE or beacons, just hoping to find some way to be able to track multiple Android phones and make it scalable.

DTTerastar, 2022-01-17:
I thought Eddystone-UID was working, i may have broken it, will check.  I will add AltBeacon.

begunfx, 2022-01-17:
@DTTerastar Eddystone-UID does show up, but it seems to be missing some ids in the payload.  If you can pass the ids in the payload then I think we can identify multiple Android phones (PS. I'm running Android 12).  Also, I seem to be getting AltBeacon as well, but again none of the ids are in the payload so same issue.  Thanks for all your work on this!

DTTerastar, 2022-01-17:
I believe the latest version adds support for both of these beacon types.  Can this be closed?

begunfx, 2022-01-17:
@DTTerastar not sure why, but when you added official support for Eddystone-UID and AltBeacon I cannot see my Android phones at all now.  Before Eddystone-UID was appearing as "sd:" entry but now it's not appearing at all (same issue for AltBeacon).

begunfx, 2022-01-18:
Eddystone-UID and AltBeacon support added and confirmed to work.

closed 2022-01-18 by begunfx