ESPresense/ESPresense #192: Getting the name of Bluetooth devices - by pairing them or within the MQTT-messages

binderth, 2022-01-19:
**Is your feature request related to a problem? Please describe.**
As mentioned in #182 I also run ESPresense without home-assistant. I _could_ install a Beacon Simulation App on each of the phones, but this seems a bit too much.

**Describe the solution you'd like**
What I'd like to have is to have ESP32 be able to pair with multiple BLE-devices. As ESP32 supports up to 9 Bluetooth-devices, I think this will do for most use cases (we're a household of four, so 9 devices are enough). 
So, I'd like to have devices paired with the ESP(s), so I can give them speaking names, which are then part of the MQTT-messages for /room and /device, so I can track in which room a device is right now and which devices are in a room right now.

**Describe alternatives you've considered**
I'm considering using a 3rd party App to just broadcast like a beacon not the right choice for mobile phone tracking. So I'd rather not go that direction.
Alternatively, it would be nice to have the "name" of the Bluetooth-device in the MQTT-message. Presently and strangely enough I don't get the name for my Android-devices, but get the name for other devices like my BLE-earpods or my Bluetooth-Victron Energy devices or even my Laptop...?

**Additional context**
So, it would be "easy" as follows:
1. search for device from ESPresense configuration page
2. pair a device with the ESP32 and name it
3. repeat for each "room" ESP32
4. get MQTT-topcis and MQTT-messages containing the BT-name to be able to filter the relevant ones out

DTTerastar, 2022-01-19:
Bluetooth Classic is very eager to give up a name, but BTLE typically isn't (ESPresence will show it if it is tho).  The 9 device limit of the ESP32 stack is really more like 3, and since you can't have all your nodes all be connected at the same name it doesn't really work.  Just connecting for a second, and querying a device has all sorts of issues...

binderth, 2022-01-19:
Oh - that's a pity. So, you mean normally an Android phone will connect via BLE and not give in its name, and "classic" devices without Low Energy will transmit their names? There goes my other option... 
My main use case would be to tell, if someone is within a specific room, and at least Samsung phones will put in "samsung" in their id, so there's topics like: `espresense/devices/samsung:559ecf52a409/keller` (with everchanging MAC spoofing), so at least I know, its a phone in that room, which helps. Would be nice to know, who is that, but yeah... ;) Heating regulation works without it...

closed 2022-01-20 by binderth