ESPresense/ESPresense #194: Apple Watch Disappears

nmajin, 2022-01-21:
**Describe the bug**
I was seeing my Apple Watch S7 go into "not_home" in Home Assistant and figured it was just my max distance that might have been off, but I found this to be odd since I have 4 ESPresense devices in a very small space. But MQTT Explorer actually shows the device missing and it eventually comes back. 

**Expected behavior**
Device to be publishing messages

**Logs**
N/A

**Hardware Details**
I am using an Apple Watch S7 and I've set it up very basic according to the procedure. I have 4 ESPresens ESP32 D1 Mini's around my house and they are calibrated very close to there should be no dead spots.

**Screenshots**
[image]

DTTerastar, 2022-01-24:
Can you capture the output of ESPresense when this happens?  There are some definite bugs in the BTLE stack of the ESP32.  I'm wondering what error code you're getting when this occurs.

nmajin, 2022-01-28:
Any idea how to capture this "output"? I assume I would need to plug the device into my laptop via USB, but then not sure if that would historically provide any details. I am also accessing the "office" device, which is the particular ESPresense that I think dropped most recently:

[image]

I know for a fact that I am providing enough max distance for all of my devices and nothing is blocking the device from getting the advertisements from my watch.

DTTerastar, 2022-01-28:
If you could use this page: https://espresense.com/terminal to capture the logs when this occurs it would be most helpful in tracking down.

duceduc, 2022-02-04:
I am experiencing the same thing on my apple watch 3. I have capture the logs from the time it shows home until not_home. The uuid is `apple:watch3-3`

    1 Del | MAC: 2b5ce5990b3c, ID: msft:cdp:0902 a8e6d0d79f42e15b71efceb37b7b88350758d48c
    1 Del | MAC: 6d2a059eb87e, ID: apple:iphone7-2 131846bfdb
    1 Model | MAC: 5c885bb879f7, ID: apple:watch3-3 Watch3,3
    lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
    0 Close | MAC: 5c885bb879f7, ID: apple:watch3-3 (0.21m) -35dBm
    0 New | MAC: 5c885bb879f7, ID: apple:watch3-3 0618810cc1
    [E][main.cpp:205] onMqttDisconnect(): Disconnected from MQTT; reason 0

    Reconnecting to WiFi...
    Connecting to WiFi SSID 'prune_dwarf'...............................192.168.1.164
    Reconnecting to MQTT...
    Reconnecting to MQTT...
    E NimBLEClient: "Connection failed; status=13 "
    1 QryErr| MAC: 604a77344b0e, ID: apple:1005:9-24 rssi -34, try 18, retry after 60000ms
    1 QryErr| MAC: acbc3268de57, ID: apple:1005:9-12 rssi -72, try 18, retry after 60000ms
    0 New | MAC: 3da6e24ecf92, ID: msft:cdp:0902 1220172f0569f3c4eff81ee217ddd1015d2f9f26
    1 Del | MAC: 112be1f62a1a, ID: exp:20 a52bb273a2bd1d0e81aa392a3805c3abc789a19d
    E NimBLEClient: "Connection failed; status=13 "
    1 QryErr| MAC: 604a77344b0e, ID: apple:1005:9-24 rssi -34, try 19, retry after 60000ms
    lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
    E NimBLEClient: "Connection failed; status=574 "
    1 QryErr| MAC: acbc3268de57, ID: apple:1005:9-12 rssi -68, try 19, retry after 60000ms
    lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
    E NimBLEClient: "Connection failed; status=574 "
    1 QryErr| MAC: 65d091d23f61, ID: apple:1005:9-7 rssi -79, try 0, retry after 1ms
    1 Model | MAC: 65d091d23f61, ID: apple:iphone12-8 iPhone12,8
    0 New | MAC: 65d091d23f61, ID: apple:iphone12-8 0318b1261f
    [E][main.cpp:205] onMqttDisconnect(): Disconnected from MQTT; reason 0

    Reconnecting to WiFi...
    Connecting to WiFi SSID 'prune_dwarf'...............................192.168.1.164
    Reconnecting to MQTT...
    Reconnecting to MQTT...
    1 Del | MAC: 7f7186815216, ID: apple:iphone7-1 031849f8b4
    1 Del | MAC: 5309a6336d42, ID: iBeacon:1ca92e23-f087-4df7-b9a2-fd4b716a4bf6-244-0 
    E NimBLEClient: "Connection failed; status=13 "
    1 QryErr| MAC: 604a77344b0e, ID: apple:1005:9-24 rssi -34, try 20, retry after 60000ms
    lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
    E NimBLEClient: "Connection failed; status=574 "
    1 QryErr| MAC: acbc3268de57, ID: apple:1005:9-12 rssi -68, try 20, retry after 60000ms
    0 New | MAC: 3dfaf217fe1e, ID: msft:cdp:0902 81c360bead17dd833083aebb36b60a5f6061e864
    1 Del | MAC: 049e9f62d14a, ID: msft:cdp:0902 acb52d5b9dbc895cebebed332597e805b64edac3
    E NimBLEClient: "Connection failed; status=13 "
    1 QryErr| MAC: 604a77344b0e, ID: apple:1005:9-24 rssi -34, try 21, retry after 60000ms
    1 QryErr| MAC: acbc3268de57, ID: apple:1005:9-12 rssi -69, try 21, retry after 60000ms
    0 Left | MAC: 76578ee2735a, ID: apple:iphone10-3 (0.21m) -80dBm
    0 Close | MAC: 76578ee2735a, ID: apple:iphone10-3 (0.22m) -38dBm
    CORRUPT HEAP: Bad head at 0x3ffcc9c8. Expected 0xabba1234 got 0x3ffcca4c
    abort() was called at PC 0x400874d1 on core 0

    ELF file SHA256: 0000000000000000

    Backtrace: 0x4008fee8:0x3ffe2450 0x40090165:0x3ffe2470 0x400874d1:0x3ffe2490 0x400875fd:0x3ffe24c0 0x401282bb:0x3ffe24e0 0x40121d61:0x3ffe27a0 0x40121ccd:0x3ffe27f0 0x40094ec1:0x3ffe2820 0x40082662:0x3ffe2840 0x400873c9:0x3ffe2860 0x4000bec7:0x3ffe2880 0x401aba9d:0x3ffe28a0 0x400e09a6:0x3ffe28c0 0x400e0b79:0x3ffe28f0 0x400e4026:0x3ffe2930 0x400e8f01:0x3ffe29a0 0x400e8fad:0x3ffe29e0 0x400e91f9:0x3ffe2a00 0x400e7ab1:0x3ffe2a20 0x400eec22:0x3ffe2a40 0x400e019b:0x3ffe2a60 0x40091176:0x3ffe2a80

    Rebooting...
    ets Jun 8 2016 00:22:57

    rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    configsip: 0, SPIWP:0xee
    clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
    mode:DIO, clock div:2
    load:0x3fff0018,len:4
    load:0x3fff001c,len:1044
    load:0x40078000,len:8896
    load:0x40080400,len:5828
    entry 0x400806ac
    Connecting to WiFi (78:E3:6D:10:0D:B0)...
    Connecting to WiFi SSID 'prune_dwarf'.......................192.168.1.164
[... 92 lines cut ...]
    [E][main.cpp:205] onMqttDisconnect(): Disconnected from MQTT; reason 0

    Reconnecting to WiFi...
    Connecting to WiFi SSID 'prune_dwarf'.............E NimBLEClient: "Connect timeout - no response"
    1 QryErr| MAC: 65d091d23f61, ID: apple:1005:9-7 rssi -79, try 0, retry after 1ms
    ..............192.168.1.164
    Reconnecting to MQTT...
    Reconnecting to MQTT...
    E NimBLEClient: "Connection failed; status=574 "
    1 QryErr| MAC: acbc3268de57, ID: apple:1005:9-12 rssi -60, try 14, retry after 16384ms
    E NimBLEClient: "Connect timeout - no response"
    1 QryErr| MAC: 5c885bb879f7, ID: apple:1005:9-24 rssi -35, try 14, retry after 16384ms
    1 Model | MAC: 65d091d23f61, ID: apple:iphone12-8 iPhone12,8
    0 New | MAC: e79341ccf8ea, ID: md:0059:8 
    0 New | MAC: 1cecc51bcabb, ID: msft:cdp:0902 4141eb95d012836466a2babf58bbb2fba9e11930
    0 New | MAC: 65d091d23f61, ID: apple:iphone12-8 431cb1261f
    0 New | MAC: 3da6e24ecf92, ID: msft:cdp:0902 1220172f0569f3c4eff81ee217ddd1015d2f9f26
    0 New | MAC: ed5cc31a97c6, ID: tile:ed5cc31a97c6 
    0 New | MAC: 001cfc7628ba, ID: 001cfc7628ba 
    0 New | MAC: 179b9fb4ac09, ID: msft:cdp:0902 4117f3fb5f148045298e41bd03e586dc772dcef9
    lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
    E NimBLEClient: "Connection failed; status=574 "
    1 QryErr| MAC: acbc3268de57, ID: apple:1005:9-12 rssi -61, try 15, retry after 32768ms
    E NimBLEClient: "Connect timeout - no response"
    1 QryErr| MAC: 5c885bb879f7, ID: apple:1005:9-24 rssi -35, try 15, retry after 32768ms

1Humza, 2022-02-05:
Same issue, except it slows down more and more and I think eventually crashes. I have query devices on.
```
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: 6c4a851a6c83, ID: apple:1005:9-12                                              rssi -82, try 13, retry after 8192ms
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 68f93e81cff5, ID: apple:1005:9-26                                              rssi -62, try 13, retry after 8192ms
scan_evt timeout
0 New   | MAC: 12103c0c30e9, ID: exp:20                                                       02101dff88d4b6ac78b4c6ed25affae841e55211
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 449e39f98a06, ID: apple:1006:10-5                                              rssi -61, try 0, retry after 1ms
E NimBLEClient: "Client busy, connected to 44:9e:39:f9:8a:06, id=0"
1 QryErr| MAC: 449e39f98a06, ID: apple:1006:10-5                                              rssi -61, try 1, retry after 2ms
E NimBLEClient: "Client busy, connected to 44:9e:39:f9:8a:06, id=0"
1 QryErr| MAC: 449e39f98a06, ID: apple:1006:10-5                                              rssi -61, try 2, retry after 4ms
E NimBLEClient: "Client busy, connected to 44:9e:39:f9:8a:06, id=0"
1 QryErr| MAC: 449e39f98a06, ID: apple:1006:10-5                                              rssi -61, try 3, retry after 8ms
E NimBLEClient: "Client busy, connected to 44:9e:39:f9:8a:06, id=0"
1 QryErr| MAC: 449e39f98a06, ID: apple:1006:10-5                                              rssi -61, try 4, retry after 16ms
E NimBLEClient: "Client busy, connected to 44:9e:39:f9:8a:06, id=0"
1 QryErr| MAC: 449e39f98a06, ID: apple:1006:10-5                                              rssi -61, try 5, retry after 32ms
```

closed 2022-04-13 by stale[bot]