ESPresense/ESPresense #198: setting "only update mqtt if beacon moved more than this distance": ignored?

wivaku, 2022-01-24:
Using v2.0.15.

I have set minimum distance change to 1 meter (and restarted the sensors):
```
Update mqtt if beacon has moved more than this distance since last report (in meters): 1
```

But I see MQTT messages where the change in distance is less than 1 meter (e.g. 2.13, 2.14, 2.12, 2.17, 2.32).

Related to https://github.com/ESPresense/ESPresense/pull/75 ?

DTTerastar, 2022-01-28:
We send based on a timer as well, because mqtt_room will mark it as away if no signal received.

wivaku, 2022-01-28:
So it is whichever comes first: either 1) distance moved or 2) time elapsed?

DTTerastar, 2022-01-28:
Correct.

wivaku, 2022-01-29:
In that case a suggestion: add an attribute to the MQTT message that shows what triggered that message. 
`trigger: time` or `trigger: distance`. This so the receiver can know if the message was created because of movement or occupancy.

ralf-koe, 2022-01-29:
I like tihis idea, possibly also a `trigger: offline` could be integrated, to show, that the values are old and beacon is not longer reachable from the dedicated ESP.

wivaku, 2022-01-29:
And -thinking out loud- another suggestion: add something like `lastMoved: <timestamp>`. This so the receiver knows when the beacon was last seen moving. Same definition as above, so: when was the last time difference between distance of current and previous sample was bigger than threshold. 

For my application an iPhone that is moving all the time is more "valuable" information than an iPad that has been laying in a room all day.

wivaku, 2022-01-29:
And related to that: add a setting for a "has not moved" threshold, e.g. treat as static after the beacon has not moved after e.g. 60  minutes.

wivaku, 2022-02-04:
Will close this one as it is not a bug.
Have added feature requests:
#218 
#219

closed 2022-02-04 by wivaku