ESPresense/ESPresense #201: Not a bug, Just a strange behavior

axelct, 2022-01-25:
Hi all,
I think this is a wonderfull project. I've started to use it to track my indoor presence through iPhone 8.
I'm using room assistant companion app to get a fixed UID for my Phone 8 and I've configured sensors (presence and RSSI) with no problem in Home Assistant, but when I'm at home sensor go "in and out" continuosly (as you can see in picture), but i'm alway at home.
[image]
Is there anything I've missing? I've configured espresence qith query and active scan enabled but nothing is changing. 
I think is s iPhone problem but I tought that with room assistant app this didn't happen.
Is anyone can help me?
Thanks in advance.

_Originally posted by @axelct in https://github.com/ESPresense/ESPresense/discussions/191_

axelct, 2022-01-25:
Thanks @DTTerastar for your answer. YEs, I'm in a zone where there are a lot of wifi. Changing maxDistance to 0 not helped. In the meanwhile I've teste Mi SmartBand 5, it works like a charm.

duceduc, 2022-01-25:
yea, I think it is an iphone issue. I am testing 2 iphones (X and 8) using the roomassistant companion app. It would work for a few hours and it go to not_home even though the phones never left. I have also tried using passive detecting and that seems to be working for a few days, but now it has detected not_home for a few hours now. I think has changed to a different uuid.

axelct, 2022-01-26:
Update: after an HA reboot (and mxDistance to 0) and after 1 day, the situation seems to be stabilized. iPhone now is always at home (if I'm at home). Thanks again @DTTerastar

closed 2022-01-28 by DTTerastar