ESPresense/ESPresense #202: Mi Flora support

vk496, 2022-01-25:
**Is your feature request related to a problem? Please describe.**
Currently I have some espresense devices arround the house, which makes great job. And I have some Mi Flora devices outside the house too.

It would be great that Espresense could detect the info that they are constantly broadcasting, so I would not need 2 esp32 (1 for presense and other for Mi Flora sensor)

thehijacker, 2022-03-27:
Hello,

Another vote for Mi Flora support. Currently I am switching all my ESP32 with Tasmota (BLE) for ESPresense as I also plan to do room presense detection. So far I have used my ESP32 with Tasmota only to collect data from Xiaomi BLE Temperature sensors (with ATC firmware) and data from the Mi Flora devices.

Now with ESPresense I noticed that it is also collection data from my temperature sensors which is great. Only thing missing is to collect from the Mi Flora also. Here is an example mqtt that Tasmota sends for the Mi Flora devices:

`{"Time":"2022-03-27T19:20:12","alias":"Flora1","mac":"c47c8d6cbd96","Temperature":24.4,"Illuminance":353,"Moisture":0,"Fertility":0,"RSSI":-91}`

Thank you great job!

DTTerastar, 2022-04-02:
Bought one on banggood...

1technophile, 2022-04-06:
You could use [Theengs decoder](https://github.com/theengs/decoder) to extend the list of the devices decoded drastically.

DTTerastar, 2022-04-09:
Cool.

closed 2022-09-07 by stale[bot]