ESPresense/ESPresense #204: Feature Request: Separate Room Status from Room Telemetry Data

begunfx, 2022-01-26:
**Is your feature request related to a problem? Please describe.**
Right now the ESP32 sensors are sending a ton of data, and are overloading my MQTT server.  If I turn off the Room topic my traffic is reduced by 50%.  The consequence of doing this is I lose any of the data pertaining to the status of my room.  

**Describe the solution you'd like**
I would like to have the Status, IP, and Version data be put in a Room Status Topic and the rest in Telemetry so I can disable telemetry broadcasts, since the device topics have the same data and reduce overall traffic, while still have the ability to track the status of each ESP32 sensor.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
There is a lot of "noise" with the sensors and any way we can clean up how the data is being broadcast so it's more efficient would improve the overall quality of this project.

DTTerastar, 2022-03-11:
I'm not sure what you're asking.  I think telemetry is already a separate option.  Does that not work?  Also the new include/exclude functionality should clean up what sensors you see so this might be unnecessary anyways.

begunfx, 2022-03-11:
To clarify a bit more.  Right now the Status is located in "Room" but the IP address, and version # are located in Telemetry.  This is not about capturing data for the beacons, but rather the sensors as I have the sensors as devices in HomeSeer so I can check their status, and create events to respond to their status.  The problem is some of the values I want are in Room and some of the values I want are in Telemetry.  I was wondering if you could put Status, IP address and Version # in Room, so I can disable Telemetry?

As mentioned above, this request is to monitor the sensors and not about the beacons. 

Hope that clarify's things a bit more.

DTTerastar, 2022-03-11:
If you don't care about beacons just put ZZZZ into include and you won't have anything but sensors and telemetry, right?

begunfx, 2022-03-11:
I do care about the beacons.  I want to monitor both.  I just wanted a way to limit what data I was monitoring for the sensors a bit better.

begunfx, 2022-03-11:
Maybe I'm not explaining myself well enough.  I'm not referring to the room information from within the beacons.  I'm referring to the rooms topic from the sensors.  Also, there is data that lives only within telemetry topic that I would like available in the room topic: "IP address" and "Version".  Hope that makes more sense.

Thanks.

closed 2022-05-18 by stale[bot]