ESPresense/ESPresense #205: Feature Request: Setting to control how often the sensor scans for devices

begunfx, 2022-01-26:
**Is your feature request related to a problem? Please describe.**
Even with the Query more often disabled, there is a LOT of traffic/noise coming from the ESP32 sensors.  

**Describe the solution you'd like**
Can we have a setting to control how often the sensors scan for devices?  I bet we can still get accurate results when reducing the scan time to 2-3 seconds, which would greatly reduce the amount of MQTT traffic.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

hvorragend, 2022-03-04:
That would be really great. 
My MQTT broker and so my home assistant sensors are really busy. A lot of traffic.

DTTerastar, 2022-03-04:
ESPresense doesn't scan, it passively listens, so there is no such setting.  You can tweak what and when to send already.  Not sure how to help with this.

begunfx, 2022-03-04:
The idea behind this request was to be able to mitigate the large volume of traffic that is sent to the MQTT broker from all the sensors.  However, since you now have the ability to whitelist or blacklist beacons, I think that solves the issue and makes this request no longer necessary.  I can close it.  Thanks!

closed 2022-03-04 by begunfx

DTTerastar, 2022-03-04:
Good to hear. Thanks.