ESPresense/ESPresense #206: Feature Request: Ability to exclude devices from future scanning

begunfx, 2022-01-26:
**Is your feature request related to a problem? Please describe.**
The list of Bluetooth devices can be really large.   Having a large list of devices to scan and report to MQTT broker can contribute to slowing down the MQTT broker.

**Describe the solution you'd like**
I would like to see an option in the sensors that lists all the detected devices (dynamically updated) that I can uncheck which ones I don't want broadcast to the MQTT broker.  That way I can limit what is sent to the broker to the devices that I care about.

soloam, 2022-02-09:
We can go that way, or simpler would be a white list, that if filled only shows that devices.

begunfx, 2022-02-09:
Yeah.  That would work too.  Thanks!

begunfx, 2022-02-09:
I would have a checkbox that would toggle whether you use or ignore the white-listed devices.  That way if you want to add additional devices, you can discover them (disable the white list) then add the new ones to the list.

ralf-koe, 2022-02-10:
Very good, I get currently from ~20 MiFlora accross 4 ESPresence a large amount of unneccessary message to my MQTT server. That will reduce unnecessary traffic exponential :-).

begunfx, 2022-02-14:
@DTTerastar @soloam I tried excluding exp:20 from all of my beacons, but when I watch the MQTT traffic (via MQTT Explorer) I still see exp:20 reporting.

begunfx, 2022-02-14:
@DTTerastar @soloam I tried the whitelist instead of the exclude list and this is what I found:
I had to restart the ESP32 sensors for the changes to take effect - seems like just saving changes isn't enough.
Also, I had added two iBeacons and one IPhone but this is what I'm getting for traffic (none of my devices whitelisted):
[image]

DTTerastar, 2022-03-10:
You need to use include in order for the whitelisting to kick in and start filtering.  There was a bug with a trailing space allowing all, but that should be resolved.  I think this can be closed.

closed 2022-03-10 by DTTerastar