ESPresense/ESPresense #208: Android and iOS Max Distance value

nickelnine, 2022-01-28:
Hi, thanks for the great work on this. In my home I'm tracking one Pixel 5 running Android 11 using the HA app and one iPhone 12 running iOS 15.1.1 using the continuity fingerprint. I'm struggling to land on a common max distance value for both phones to use. When setting a max distance of 1.5 for example in my `livingroom`, I find the Pixel reports `livingroom`, but the iPhone reports `not_home` even though the iPhone is a lot closer to the beacon. Similarily, setting a value for the iPhone, the Pixel doesn't like it and reports `not_home`. I've spent a lot of time tweaking the max distance but cannot figure how to satisfy both devices. Is there a way to split max distance for each phone? or is there another setting I'm not utilizing?

Another observation, when watching mqtt, the iPhone I want to track - apple:iphone13-3 - is reporting `livingroom` when the device is actually in the `bedroom` and iPhones that are not in the house at all are reporting `bedroom`. Here is an example:
`livingroom = {"id":"apple:iphone13-3","name":"iPhone13,3","disc":"710c0f05803591bd9f","idType":40,"rssi@1m":-65,"rssi":-74,"mac":"660152b33d2e","raw":1.81,"distance":1.93,"speed":0}`
`bedroom = {"id":"apple:iphone12-1","name":"iPhone12,1","disc":"331af4c12075","idType":40,"rssi@1m":-99,"rssi":-93,"mac":"64c9bc332d88","raw":0.67,"distance":0.41,"speed":0}`

I'm really at a loss and would appreciate the feedback. Thanks again

nickelnine, 2022-01-28:
User error

nickelnine, 2022-01-31:
Hi, thanks for the great work on this. I'm tracking a Pixel 5 running Android 11 and an iPhone 12 Pro running iOS 15 and I'm struggling with a couple of things:

1.  The max distance value for each phone varies a lot for each room I'm trying to track movement in. For example, in my livingroom, my Pixel works with a max distance of 1.3 for the boundary I have set but the iPhone is always reporting not_home until I raise the distance to say 2 then the iPhone is satisfied and reports livingroom but then the Pixel will start reporting livingroom when it is outside the boundary, so it seems I cannot nail a common max distance between each phone, one distance is disliked by one of the other phones. Given both phones run different BLE stacks (I'm not very knowledgeable in the field) is there anything I can do to help fix this? Any setting I'm overlooking?
2.  Before posting this comment, I reflashed my livingroom ESP32 to see if it would help resolve the above and I found that the iPhone 12 Pro was not logging at all in the terminal even running the terminal session for an hour. I should be seeing "apple:iPhone13,3" (or similar)  but instead I was seeing all types of other iPhones and devices logged presumably from iPhones in my neighborhood. I also tried the room-assistant app but no luck there either. Is there anything I can do in this case? (besides forcing the wife to switch to Android?)

Thanks for any input and feedback. It is mildly frustrating but all part of the game I suppose.

nickelnine, 2022-02-01:
Messing around with the settings in the HA companion app as well as ESPresense, I'm doing better with tracking than before but I'm still getting the iPhone 12 to go 'not_home' pretty quickly after getting a stable 'livingroom' for a few minutes. I am using the room-assistant app. When I increase the maxDistance and reboot, the iPhone sticks on 'livingroom' but then switches to 'not_home' soon after and doesn't come back. Any tips here?

nickelnine, 2022-02-10:
I think I got it working somewhat playing with the RSSI and max distance together.

closed 2022-02-10 by nickelnine

Jazza159, 2023-02-13:
Hey Nick,

I have a similar issue and would love your input on how you got it working more reliably.
My Samsung S-22 will report 10 m away from a base station and the iPhone will report 1-2m away. It is changing between rooms very frequently as the difference between them is centimetres.

In the HA companion app on the S22 I have the BLE strength set to the highest setting but there is still a massive difference.

Did you find a way to get them to report similar numbers or maybe filter RSSI for each device individually?
Cheers!

nickelnine, 2023-02-23:
I fiddled a lot with the Android HA companion app BLE settings combination of Advertise mode, Measured power and Transmitter power. Balanced, -89 and High in my case.
As for iOS, I enabled IRK and set the max distance of -85 in the web GUI of Espresense of each base station.

It took a lot of trial error and patience to get solid readings and reliable room detection. I sat with each phone in my hand on the perimeter of each room to get the numbers I did as well as some educated guesses.

I hope that helps, I know it's not much but you just have to hack away at it I'm afraid.