ESPresense/ESPresense #209: A question about supported BLE Devices

jeremyoha450, 2022-01-28:
Is there any devices with BLE work with this?
like any particular watch, fit device, etc?

regards
Jeremy

DTTerastar, 2022-01-28:
https://espresense.com/beacons

mahert45, 2022-02-12:
I have a Garmin Venu SQ and it works but only when not connected to my phone. 

Is this normal behavior?

tonyyng, 2022-02-13:
I see the same behavior with my Fitbit Charge 4.  If the watch is near my phone, it stops being detected by the ESP32.

DTTerastar, 2022-02-14:
We are basically "eavesdropping" on these devices to listen for messages they're sending.  Some devices send less advertisements when connected, or nearby the device it's looking for.  The Apple Watch however is always sending out messages regardless.

ozett, 2022-03-01:
> https://espresense.com/beacons

@DTTerastar : i search now the issues for tiles that i can track without intervention for their distance.
the meantioned list is somehow short to help me find working tiles.

is there a general rule for key-finder tiles on battery (bt5/LE)? 
or do i have to avoid bt4 and try lots of stuff from ali? 
maybe some send advertisments, and some not? 

this list is massive:
https://github.com/custom-components/ble_monitor

can they all be tracked to?
https://custom-components.github.io/ble_monitor/faq#how-can-i-use-my-device-as-a-tracker

DTTerastar, 2022-03-02:
Most all the ble_monitor we support as well.  I don't have everyone so I can't possibly vouch for them all.  But both work similarly.

ozett, 2022-03-02:
Thanks @DTTerastar 
i ordered some different tiles on ali and will see if they advertise themselves.

is there a general rule, which device sends advertisments in which intervals?
only from BT 5.0 upwards?

in my box i found an old key-chain tile with BT 4.0, and it sends only packets on button-press.
this is not usable to automatically determine the approaching distance of the tile. 
Hope that the BT5 tiles will constantly announce themselves.. maybe i invest in an expensive "Tile"-tile to experiment?

ozett, 2022-03-02:
looks like i have to find BLE (5+)? that are "broadcaster" or "advertiser" devices.
which sends in short intervals, not once a day

[image]

[image]

ozett, 2022-03-04:
looks like there are beacons which can be configured 
when to send
and what to announce

https://reelyactive.github.io/diy/minew-e8-config/

[image]

DTTerastar, 2022-03-04:
ESPHome can also listen to advertisements.  The point of this project is to fingerprint things w/ rotating macs like apple devices.  But we also support anything w/ a public mac address.

ozett, 2022-03-05:
i am reading some basics about BLE advertising to get a little deeper knowledge.

for my use case, which is identify a bluetooth-device (tile,tracker..) near to the frontdoor and open door,
the advertising interval seems also critical. 

fingerprinting is very helpful to identify each device,
advertising interval seems important for quick interaction to a device. 
therefore minew seems to have some interesting devices on their list for tracking moving BLE-Tiles...
maybe i am wrong and must intensify my readings to understand the mqtt-received BLE-data.. 😄

DTTerastar, 2022-03-11:
Advertising interval is important for quick location updates.  Tiles are good for figuring out what room you left something in, not for turning on the lights when moving.

ozett, 2022-03-11:
thanks for keeping on track...

i ordered some devices from china and will test them to configure advertising intervals. will report here.

i would like to get the full advertising string as json to node-red, to see if can make my own logic.
do you think that could become a feature of espresence?

DTTerastar, 2022-03-12:
The full advertising string is very easy to do via esphome.  The hard part is deciphering that into a usable fingerprint.

ozett, 2022-03-12:
sending the raw string (as an option) would be perfect for me. via mqtt
it is decipherable, as it has a known syntax.
and wourld be easy for devices without randomization (like beacons). 
putting this in sounds great ...

(give also the option for making fingerprinting with node-red logic, what would be possible after receiving the raw advertisment via MQTT..)

closed 2022-05-19 by stale[bot]