ESPresense/ESPresense #210: Manual firmware update from web GUI

TheNuttyP, 2022-01-28:
First, thank you for this product. It is amazing and much appreciated.

The WiFi that all my IoT devices connect to is firewalled off and has no access to the internet, so automatic updates are not possible. 

Would it be possible to add a feature that allows me to do a manual firmware update OTA, rather than needing to disconnect each ESP32 and connect to a computer? Then I could download the firmware and upload to each one.

As a secondary request, is there a way to have a centralized "control panel" where I could manage the settings/firmware of a house full of ESPresense sensors without needing to connect to each one individually?

Once again, your time and efforts here are very much appreciated!

DTTerastar, 2022-01-28:
You can remotely flash them OTA w/ espota as long as it's enabled.  You can do it via a shell script so you update in batch.

TheNuttyP, 2022-01-28:
I am not familiar with espota. ESPresense is my first experience with anything ESP32 related. I'll check it out. Thank you!

Can settings be get/set over the same tool, or just firmware?

DTTerastar, 2022-01-28:
Most of the "important" settings can be set via mqtt, anything specific you want added?

TheNuttyP, 2022-01-31:
Nope. Just new to ESPresense and MQTT and not sure all the details on what can be accomplished. Dug a little deeper and it looks like the settings can be updated from in Home Assistant (through MQTT), which is great! So it is just the firmware that is a little "complicated" to update OTA with it locked down like I have my IoT VLAN.

Thank you for taking the time to respond. I appreciate it.

closed 2022-02-02 by DTTerastar