ESPresense/ESPresense #211: ESP32 Panic Crash

Zipties, 2022-01-29:
**Describe the bug**
I am seeing my HiLetgo ESP32 hard crash at random times. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Config**

- [x ] Active Scan enabled
- [x ] Query enabled

**Logs**
11:56:45.730 -> 0 New   | MAC: 73fd43515a31, ID: msft:cdp:0902                                                3b7624d18ec1154b51d94f7298da9252a37b5f55
11:56:53.949 -> 0 New   | MAC: 59692097d852, ID: md:00e0:8                                                    
11:56:55.346 -> 0 New   | MAC: 5c1a4be17030, ID: md:00e0:8                                                    
11:56:56.031 -> 0 New   | MAC: 585e9d2c456d, ID: md:00e0:8                                                    
11:57:01.549 -> E NimBLEClient: "Connection failed; status=13 "
11:57:01.549 -> 1 QryErr| MAC: 63bd10ae3973, ID: apple:1005:9-24                                              rssi -88, try 14, retry after 16384ms
11:57:05.068 -> 0 New   | MAC: 553f0ad985d2, ID: md:00e0:8                                                    
11:57:06.683 -> lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
11:57:06.683 -> E NimBLEClient: "Connection failed; status=574 "
11:57:06.683 -> 1 QryErr| MAC: 6c4a8528e329, ID: apple:1005:9-12                                              rssi -90, try 53, retry after 60000ms
11:57:09.148 -> lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
11:57:09.148 -> E NimBLEClient: "Connection failed; status=574 "
11:57:09.148 -> 1 QryErr| MAC: 5952393b91f0, ID: apple:1006:10-8                                              rssi -79, try 0, retry after 1ms
11:57:09.148 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
11:57:09.245 -> Core 0 register dump:
11:57:09.245 -> PC      : 0x400efd3c  PS      : 0x00060330  A0      : 0x800eff8f  A1      : 0x3ffe3420  
11:57:09.245 -> A2      : 0x00000000  A3      : 0x00000016  A4      : 0x00000000  A5      : 0x00000000  
11:57:09.245 -> A6      : 0x3ffe35ac  A7      : 0xff000000  A8      : 0x801abaa0  A9      : 0xad567800  
11:57:09.245 -> A10     : 0x000000ba  A11     : 0x00000001  A12     : 0x00000014  A13     : 0x3f4045a8  
11:57:09.245 -> A14     : 0xad567800  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
11:57:09.245 -> EXCVADDR: 0xad567800  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
11:57:09.245 -> 
11:57:09.245 -> ELF file SHA256: 0000000000000000
11:57:09.245 -> 
11:57:09.245 -> Backtrace: 0x400efd3c:0x3ffe3420 0x400eff8c:0x3ffe3440 0x400f541d:0x3ffe3460 0x400f6a1e:0x3ffe3670 0x400d2680:0x3ffe3690 0x400e0b61:0x3ffe36b0 0x400e4026:0x3ffe36f0 0x400e8f01:0x3ffe3760 0x400e8fad:0x3ffe37a0 0x400e91f9:0x3ffe37c0 0x400e7ab1:0x3ffe37e0 0x400eec22:0x3ffe3800 0x400e019b:0x3ffe3820 0x40091176:0x3ffe3840
11:57:09.245 -> 
11:57:09.245 -> Rebooting...
11:57:09.245 -> ets Jun  8 2016 00:22:57

**Screenshots**
If applicable, add screenshots to help explain your problem.

duceduc, 2022-02-04:
yes, I am experiencing this myself. It crashes and never recovered from a reboot. Have to manually unplug.

RaoulSargent, 2022-02-06:
yes me too.
I have 5x esp32 running v2.0.15 suffering random restarts.
ranging from only after a couple of minutes to a couple of hours.

also had the state where after a couple of days the device no longer responds to network, ping, etc, and shows "offline" in MQTT.
This state requires a physical unplug/replug to bring the device back online.

Also had a situation where even unplug/replug (several times) did not revive the device, and I had to re-flash the esp32 to get things working again.   This was done retaining settings (not a full wipe) and worked as normal again after refreshing with same v2.0.15

Zipties, 2022-02-07:
> Also had a situation where even unplug/replug (several times) did not revive the device, and I had to re-flash the esp32 to get things working again. This was done retaining settings (not a full wipe) and worked as normal again after refreshing with same v2.0.15

Mine get slow to see devices after a few hours up until they become unresponsive. I think they are running into memory issues. I have a convoluted automation setup to power cycle a zigbee plug they are on every hour. At least they reboot fast

RaoulSargent, 2022-02-07:
> Mine get slow to see devices after a few hours up until they become unresponsive. I think they are running into memory issues. I have a convoluted automation setup to power cycle a zigbee plug they [...]

Likewise. I have a Node-Red flow to restart all 5x espresense every 1 hour, if they have stayed up that long.
Yes, feels like a memory type issue, although not sure that would be the cause of a crash/restart after only a couple of minutes which I sometimes get.   So could it be related to what devices it has seen perhaps, and not handling those devices correctly, just a thought.

I will try and make time to start diagnosing logically rather than speculating.

Zipties, 2022-02-07:
> although not sure that would be the cause of a crash/restart after only a couple of minutes which I sometimes get. 

I get that as well, typically with query turned on. But, I don't believe at least in my case, it is related to this bug report.

RaoulSargent, 2022-02-07:
My Settings (same on 5x v2.0.15):

**Preferences:**
- Status LED = **YES**
- Automatic Update = no
- Arduino OTA update = no
- Home Assistant discovery = no
- Active Scanning = **YES**
- Send to telemetry topic = **YES**
- Send to rooms topic = **YES**
- Send to devices topic = **YES**

**Calibration**:
- Maximum distance = **2m**
- Forget beacon if not seen for = blank/default
- RSSI expected at 1m = **-70**

**Other Sensors**:
- None in use currently

wabit, 2022-02-09:
I am also seeing the same issue, but I'm lucky if it last 5 mins before crashing

RaoulSargent, 2022-02-09:
Setup a v2.0.15 with the Verbose logging firmware.
Just had a crash

Summary Previous:
Powered on 15:20
Restarted at 15:48
Crashed 15:50 so had only been running for a couple of minutes when crashed as follows...

`15:50:19.162 -> Verbose | 7e0d4687a027                                              MD: 4c0010050a18101414
15:50:19.162 -> CORRUPT HEAP: Bad head at 0x3ffe8814. Expected 0xabba1234 got 0x3ffe8df0
15:50:19.196 -> abort() was called at PC 0x400874d1 on core 0
15:50:19.196 -> 
15:50:19.196 -> ELF file SHA256: 0000000000000000
15:50:19.196 -> 
15:50:19.196 -> Backtrace: 0x4008fee8:0x3ffe2990 0x40090165:0x3ffe29b0 0x400874d1:0x3ffe29d0 0x400875fd:0x3ffe2a00 0x401288db:0x3ffe2a20 0x40122381:0x3ffe2ce0 0x401222ed:0x3ffe2d30 0x40094ec1:0x3ffe2d60 0x40082662:0x3ffe2d80 0x400873c9:0x3ffe2da0 0x4000bec7:0x3ffe2dc0 0x401ac131:0x3ffe2de0 0x400e0ab2:0x3ffe2e00 0x400e0c85:0x3ffe2e30 0x400e413e:0x3ffe2e70 0x400e9019:0x3ffe2ee0 0x400e90c5:0x3ffe2f20 0x400e9311:0x3ffe2f40 0x400e7bc9:0x3ffe2f60 0x400eed3a:0x3ffe2f80 0x400e0293:0x3ffe2fa0 0x40091176:0x3ffe2fc0
15:50:19.231 -> 
15:50:19.231 -> Rebooting...
15:50:19.231 -> ets Jun  8 2016 00:22:57`

RaoulSargent, 2022-02-09:
Same esp32, Another example:

`16:20:21.810 -> Verbose | apple:1005:9-4                                            MD: 4c0010050110fccb7d
16:20:21.810 -> CORRUPT HEAP: Bad head at 0x3ffe9cf8. Expected 0xabba1234 got 0x3ffea104
16:20:21.843 -> abort() was called at PC 0x400874d1 on core 1
16:20:21.843 -> 
16:20:21.843 -> ELF file SHA256: 0000000000000000
16:20:21.843 -> 
16:20:21.843 -> Backtrace: 0x4008fee8:0x3ffdd9e0 0x40090165:0x3ffdda00 0x400874d1:0x3ffdda20 0x400875fd:0x3ffdda50 0x401288db:0x3ffdda70 0x40122381:0x3ffddd30 0x401222ed:0x3ffddd80 0x40094ec1:0x3ffdddb0 0x40082662:0x3ffdddd0 0x400873c9:0x3ffdddf0 0x4000bec7:0x3ffdde10 0x401ac131:0x3ffdde30 0x400e0819:0x3ffdde50 0x400e09ba:0x3ffdde70 0x400f15df:0x3ffddea0 0x400f7037:0x3ffddf10 0x400d7769:0x3ffde0c0 0x40091176:0x3ffde140
16:20:21.876 -> 
16:20:21.876 -> Rebooting...
16:20:21.876 -> ets Jun  8 2016 00:22:57
16:20:21.876 -> 
16:20:21.876 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
`

RaoulSargent, 2022-02-09:
```18:24:29.917 -> Verbose | 7ac88defa2aa                                              MD: 4c0010050b182b2484
18:24:29.917 -> CORRUPT HEAP: Bad head at 0x3ffe96f4. Expected 0xabba1234 got 0x3ffe9880
18:24:29.917 -> abort() was called at PC 0x400874d1 on core 1
18:24:29.917 -> 
18:24:29.917 -> ELF file SHA256: 0000000000000000
18:24:29.917 -> 
18:24:29.917 -> Backtrace: 0x4008fee8:0x3ffdd900 0x40090165:0x3ffdd920 0x400874d1:0x3ffdd940 0x400875fd:0x3ffdd970 0x401288db:0x3ffdd990 0x40122381:0x3ffddc50 0x401222ed:0x3ffddca0 0x40094ec1:0x3ffddcd0 0x40082662:0x3ffddcf0 0x400873c9:0x3ffddd10 0x4000bec7:0x3ffddd30 0x401ac131:0x3ffddd50 0x400e0819:0x3ffddd70 0x400e09ba:0x3ffddd90 0x400f15df:0x3ffdddc0 0x400f7037:0x3ffdde30 0x400d7769:0x3ffddfe0 0x40091176:0x3ffde060
18:24:29.950 -> 
18:24:29.950 -> Rebooting...
18:24:29.950 -> ets Jun  8 2016 00:22:57
18:24:29.950 -> 
18:24:29.950 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

RaoulSargent, 2022-02-09:
This is a great project, so I would like to do whatever I can to assist with debugging/testing this issue... just let me know if I can help.

wabit, 2022-02-09:
I have done a bit more testing and it happens a lot more if I flash direct from the site compared to doing it through espflasher. When doing it through the site I rarely get to the point of it connecting to Wi-Fi before a hard crash. 

I'm going to try a higher end usb cable and a brand new esp32 using espflasher and see if there is a change. 

With how bad it is for me it must be something on my side otherwise it would be unusable for everyone and going of discord and Reddit that's not the case.

DTTerastar, 2022-02-09:
Platform.Io can "decode" the stack dumps automatically so we can see where the error is occuring.  It is enabled here: https://github.com/ESPresense/ESPresense/blob/master/platformio.ini#L39

So in order to use it you need to run it via `pio run -t upload -t monitor`.  Also to investigate if memory is running out you can use the /telemetry topic like this:

- platform: mqtt
  state_topic: espresense/rooms/family/telemetry
  value_template: '{{ value_json.maxAllocHeap }}'
  name: 'ESPresense Family Memory'

RaoulSargent, 2022-02-10:
I don't have PlatformIO installed, so first I started watching the telemetry values:

_So far I do not think I see any obvious memory leak patterns that correlate with the points when the devices crash._

Attached screenshot image shows the MQTT traces for both devices.
Note: Devices are restarted by Node-Red should they still be running after 1 hour.

Both Devices are in the same room, approx 30cm apart, so should discover same beacons.
Both powered off and on at the same time.
(Non-default settings)
- Max distance = 2
- Forget Beacon = 120000
- Update MQTT = 30000

**The MQTT payloads below are the payload of the last message prior to the crash**

**DEVICE A:**    v2.0.15
09:12:00   Power Off/On
09:27:48   CRASH A1
{"ip":"192.168.14.107","uptime":927,"firm":"esp32","rssi":-79,"ver":"v2.0.15","adverts":23066,"seen":23030,"queried":67,"reported":245,"freeHeap":126244,"minFreeHeap":115860,"maxAllocHeap":85716,"resetReason":"PowerOn"}

10:28:12   RESTARTED BY NODE-RED

10:45:13   CRASH

11:15:07    CRASH

11:20:56   CRASH

11:41:11   CRASH
{"ip":"192.168.14.107","uptime":1209,"firm":"esp32","rssi":-77,"ver":"v2.0.15","adverts":21015,"seen":20974,"queried":112,"reported":372,"freeHeap":125812,"minFreeHeap":116688,"maxAllocHeap":87884,"resetReason":"SoftwareCpu"}

**DEVICE B:**     v2.0.15 Verbose
09:12:00   Power Off/On
09:58:37   {"ip":"192.168.14.251","uptime":2777,"firm":"esp32-verbose","rssi":-79,"ver":"v2.0.15","adverts":65458,"seen":65403,"queried":127,"reported":845,"freeHeap":129228,"minFreeHeap":108644,"maxAllocHeap":90568,"resetReason":"PowerOn"}
09:58:37.538 -> CORRUPT HEAP: Bad head at 0x3ffe9350. Expected 0xabba1234 got 0x3ffe94dc
09:58:37.538 -> abort() was called at PC 0x400874d1 on core 1
09:58:37.538 -> 
09:58:37.538 -> ELF file SHA256: 0000000000000000
09:58:37.538 -> 
09:58:37.538 -> Backtrace: 0x4008fee8:0x3ffdd980 0x40090165:0x3ffdd9a0 0x400874d1:0x3ffdd9c0 0x400875fd:0x3ffdd9f0 0x401288db:0x3ffdda10 0x40122381:0x3ffddcd0 0x401222ed:0x3ffddd20 0x40094ec1:0x3ffddd50 0x40082662:0x3ffddd70 0x400873c9:0x3ffddd90 0x4000bec7:0x3ffdddb0 0x401ac131:0x3ffdddd0 0x400e0819:0x3ffdddf0 0x400e09ba:0x3ffdde10 0x400f15df:0x3ffdde40 0x400f7037:0x3ffddeb0 0x400d7769:0x3ffde060 0x40091176:0x3ffde0e0
09:58:37.608 -> 
09:58:37.608 -> Rebooting...
09:58:37.608 -> ets Jun  8 2016 00:22:57
09:58:37.608 -> 
09:58:37.608 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

10:58:59   RESTARTED BY NODE-RED

11:34:08   CRASH
{"ip":"192.168.14.251","uptime":2111,"firm":"esp32-verbose","rssi":-76,"ver":"v2.0.15","adverts":39236,"seen":39157,"queried":72,"reported":655,"freeHeap":128972,"minFreeHeap":118664,"maxAllocHeap":91028,"resetReason":"SoftwareCpu"}

[image]

DTTerastar, 2022-02-10:
This is what it looks like with the pio debugger on:

```ELF file SHA256: 0000000000000000

Backtrace: 0x4008ffec:0x3fff2400 0x40090269:0x3fff2420 0x401a99db:0x3fff2440 0x401a9a22:0x3fff2460 0x401a934d:0x3fff2480 0x401a9424:0x3fff24a0 0x400f5f8d:0x3fff24c0 0x400d6839:0x3fff24e0 0x4009127a:0x3fff2560
  #0  0x4008ffec:0x3fff2400 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40090269:0x3fff2420 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x401a99db:0x3fff2440 in __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
  #3  0x401a9a22:0x3fff2460 in std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
  #4  0x401a934d:0x3fff2480 in __cxa_allocate_exception at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc:313
  #5  0x401a9424:0x3fff24a0 in operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
  #6  0x400f5f8d:0x3fff24c0 in __gnu_cxx::new_allocator<std::_List_node<BleFingerprint*> >::allocate(unsigned int, void const*) at /Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ext/new_allocator.h:104
      (inlined by) std::__cxx11::_List_base<BleFingerprint*, std::allocator<BleFingerprint*> >::_M_get_node() at /Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_list.h:392
      (inlined by) std::_List_node<BleFingerprint*>* std::__cxx11::list<BleFingerprint*, std::allocator<BleFingerprint*> >::_M_create_node<BleFingerprint* const&>(BleFingerprint* const&) at /Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_list.h:571
      (inlined by) void std::__cxx11::list<BleFingerprint*, std::allocator<BleFingerprint*> >::_M_insert<BleFingerprint* const&>(std::_List_iterator<BleFingerprint*>, BleFingerprint* const&) at /Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_list.h:1763
      (inlined by) void std::__cxx11::list<BleFingerprint*, std::allocator<BleFingerprint*> >::emplace_back<BleFingerprint* const&>(BleFingerprint* const&) at /Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_list.h:1099
      (inlined by) void std::__cxx11::list<BleFingerprint*, std::allocator<BleFingerprint*> >::_M_initialize_dispatch<std::_List_const_iterator<BleFingerprint*> >(std::_List_const_iterator<BleFingerprint*>, std::_List_const_iterator<BleFingerprint*>, std::__false_type) at /Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_list.h:1692
      (inlined by) std::__cxx11::list<BleFingerprint*, std::allocator<BleFingerprint*> >::list(std::__cxx11::list<BleFingerprint*, std::allocator<BleFingerprint*> > const&) at /Users/dturner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_list.h:657
      (inlined by) BleFingerprintCollection::getCopy[abi:cxx11]() at lib/BleFingerprint/BleFingerprintCollection.cpp:63
  #7  0x400d6839:0x3fff24e0 in scanForDevices(void*) at .pio/libdeps/m5stickc-plus/ArduinoJson/src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15
  #8  0x4009127a:0x3fff2560 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

I think that says the list allocator ran out of memory.  I think I just may be putting too much stuff in the list of discovered beacons.

DTTerastar, 2022-02-10:
Well that was because the m5stickc-plus uses way too much memory for it's graphics routines.  I switched to the esp32 flavor and did find an issue:

```
E (401988) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (401988) task_wdt:  - IDLE0 (CPU 0)
E (401988) task_wdt: Tasks currently running:
E (401988) task_wdt: CPU 0: btController
E (401988) task_wdt: CPU 1: IDLE1
E (401988) task_wdt: Aborting.
abort() was called at PC 0x4018a23c on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f658:0x3ffbfee0 0x4008f8d5:0x3ffbff00 0x4018a23c:0x3ffbff20 0x4008dd01:0x3ffbff40 0x40190881:0x3ffb61d0 0x40190e55:0x3ffb61f0 0x400908e6:0x3ffb6220
  #0  0x4008f658:0x3ffbfee0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4008f8d5:0x3ffbff00 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x4018a23c:0x3ffbff20 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x4008dd01:0x3ffbff40 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x40190881:0x3ffb61d0 in r_rw_schedule at ??:?
  #5  0x40190e55:0x3ffb61f0 in btdm_controller_task at ??:?
  #6  0x400908e6:0x3ffb6220 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
```

DTTerastar, 2022-02-10:
Added an issue: https://github.com/h2zero/NimBLE-Arduino/issues/351

DTTerastar, 2022-02-11:
Well I think it is more memory based than I previously thought.  I base that on my machina-a0 which has more memory than all my other modules has the best uptime and free memory.  I've purchased a esp32 w/ a debugger so I can better figure out what is using all the memory.  Stay tuned.

DTTerastar, 2022-02-12:
Hmm, that is very helpful!

SchallerJe, 2022-02-12:
> Just some feedback, I flashed two identical brand new ESP32's. Both run fine on standard setting (connected to WiFi and MQTT), as soon as I enable "Query Devices" on either device then I get Panic Crash very regularly (every 2 minutes or so).
> 
> Have iPhone and Apple Watch in case this is agitated by processing the devices.

I installed ESPresense for the first time today onto an AZ-Delivery D1 Mini and have the same issue: enabling "Query Devices" almost immediately crashes the device.

Stubbs, 2022-02-14:
Some more core dumps that might be related:

```
CORRUPT HEAP: Bad head at 0x3ffe3218. Expected 0xabba1234 got 0x3ffe3444
abort() was called at PC 0x40086cdd on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f658:0x3ffe1d80 0x4008f8d5:0x3ffe1da0 0x40086cdd:0x3ffe1dc0 0x40086e09:0x3ffe1df0 0x401266db:0x3ffe1e10 0x40120181:0x3ffe20d0 0x401200ed:0x3ffe2120 0x400943ed:0x3ffe2150 0x40081e6e:0x3ffe2170 0x40086bd5:0x3ffe2190 0x4000bec7:0x3ffe21b0 0x401a9ea5:0x3ffe21d0 0x400e06fe:0x3ffe21f0 0x400e08d1:0x3ffe2220 0x400e3d96:0x3ffe2260 0x400e8c71:0x3ffe22d0 0x400e8d1d:0x3ffe2310 0x400e8f69:0x3ffe2330 0x400e7821:0x3ffe2350 0x400ee992:0x3ffe2370 0x400dff63:0x3ffe2390 0x400908e6:0x3ffe23b0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```

ptgomes, 2022-02-17:
Also same problem. It crashes after 1 or 2 minutes after giving HA correct distances. Only way to reboot is unplugging it.

RaoulSargent, 2022-03-15:
> Ok, try the prerelease I think the uptime is massively improved.

Significant improvement :-)
I have 6x esp32 Wemos D1 Mini running on v2.0.38b7 and then auto-self-updated OTA to v2.1.0
Apart from when the OTA update was performed they appear to have been up and running constantly.
The longest currently is 16 hours and counting, where the previous record was less than 2 hours, so this is a "significant" improvement.

Thank You :-)

closed 2022-03-22 by DTTerastar

RaoulSargent, 2022-03-24:
Here are some uptime charts for the 6x devices I have running.
Note still getting crashes, just not as often.

Also had a situation where the two devices in the 'Cave' stopped triggering automation, but after a power cycle the automations started working again.  At this point I have done no diagnostics other than the power-cycle.

[image]