ESPresense/ESPresense #214: 100% QryErr immediately after enabling query

Zipties, 2022-01-31:
**Describe the bug**
When Query is enabled the log immediately reports every device with an error, and stops responding to "close" and "left". The issue does not go away with power cycles. 

**Expected behavior**
occasional errors

**Hardware Details**
I have ESPresense running on 5 ESP32. One is a HiLetgo model the others are Atom M5s. I get the exact same behavior on all of them

**Config**

- [x ] Active Scan enabled (tried on and off)
- [x ] Query enabled

**Logs**
`Connected 115200 8N1...
�>��>�>�����>���>�>��>�8���8�8�8�>88��8888��8��>8���8��>�������8>����>��������>��8�E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 1, retry after 2ms
E NimBLEClient: "Client busy, connected to 65:13:7e:f9:2a:16, id=1"
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 2, retry after 4ms
E NimBLEClient: "Client busy, connected to 65:13:7e:f9:2a:16, id=1"
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 3, retry after 8ms
E NimBLEClient: "Client busy, connected to 65:13:7e:f9:2a:16, id=1"
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 4, retry after 16ms
E NimBLEClient: "Client busy, connected to 65:13:7e:f9:2a:16, id=1"
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 5, retry after 32ms
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 6, retry after 64ms
E NimBLEClient: "Client busy, connected to 65:13:7e:f9:2a:16, id=0"
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 7, retry after 128ms
1 Model | MAC: 5bae83d64d52, ID: apple:iphone13-2                                             iPhone13,2
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 8, retry after 256ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 9, retry after 512ms
0 New   | MAC: 7b455c46c3a3, ID: roomAssistant:01d573c0-8773-4179-9ce5-8d4c1dcc6f4e           391fd5fc310548
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 10, retry after 1024ms
1 Model | MAC: 51c10d4444b3, ID: apple:ipad7-3                                                iPad7,3
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 11, retry after 2048ms
0 New   | MAC: 5bae83d64d52, ID: apple:iphone13-2                                             7b1e6ef16f19
0 New   | MAC: 575ae61f08d0, ID: md:00e0:8                                                    
0 New   | MAC: beffe701084d, ID: beffe701084d                                                 
0 New   | MAC: 40c7c273b1c4, ID: msft:cdp:0902                                                962ec599cd66b350eff67e7dfb9dc3df20851abf
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: 5de06e2f3532, ID: apple:1007:11-7                                              rssi -87, try 0, retry after 1ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: 5de06e2f3532, ID: apple:1007:11-7                                              rssi -87, try 1, retry after 2ms
1 Model | MAC: 5de06e2f3532, ID: apple:iphone14-3                                             iPhone14,3
E NimBLEClient: "Connect timeout - no response"
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 12, retry after 4096ms
0 New   | MAC: 557002b8eb1a, ID: md:00e0:8                                                    
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 0, retry after 1ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 1, retry after 2ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 2, retry after 4ms
E NimBLEClient: "Connection failed; status=13 "
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 13, retry after 8192ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 3, retry after 8ms
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 4, retry after 16ms
E NimBLEClient: "Client busy, connected to c8:69:cd:15:bc:bd, id=0"
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 5, retry after 32ms
E NimBLEClient: "Client busy, connected to c8:69:cd:15:bc:bd, id=0"
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 6, retry after 64ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 7, retry after 128ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 8, retry after 256ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 9, retry after 512ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 10, retry after 1024ms
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 11, retry after 2048ms
E NimBLEClient: "Connection failed; status=13 "
1 QryErr| MAC: 65137ef92a16, ID: apple:1005:9-26                                              rssi -72, try 14, retry after 16384ms
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
1 QryErr| MAC: c869cd15bcbd, ID: apple:1005:9-12                                              rssi -75, try 12, retry after 4096ms
0 New   | MAC: 389c481bb103, ID: exp:20                                                       f9be4a6c08144f3c6e933c5fd228fb04cb0671ce
0 New   | MAC: c869cd15bcbd, ID: apple:1005:9-12                                              01143bc831`

DTTerastar, 2022-01-31:
Turning off this logging has been requested but denied: https://github.com/h2zero/NimBLE-Arduino/issues/299#issuecomment-934597330

Zipties, 2022-01-31:
> Turning off this logging has been requested but denied: [h2zero/NimBLE-Arduino#299 (comment)](https://github.com/h2zero/NimBLE-Arduino/issues/299#issuecomment-934597330)

The feature does not work at all for me though, once it is enabled all devices will time out and report not_home. Is it recommended to not use the query feature at this time?

DTTerastar, 2022-01-31:
The id changes when it's turned on.  Instead of seeing a generic id, turning on query makes it use an id that has the model number in it and/or room assistant.  In the log I see MAC: 65137ef92a16 never got a response most likely because it's too far away from the node.  But I DO see MAC: 5de06e2f3532 got a response and switched to a different id.

DTTerastar, 2022-01-31:
What is needed is a hive mind mode where all the nodes intelligently issue query commands from closer nodes, and all nodes use the new id.  Know any c++?

Zipties, 2022-01-31:
> Know any c++?

Sorry, I am just an end-user im afraid. It does not seem to be a distance issue though, pressing a device against the ESP32 will no longer trigger the "close" message after the errors start rolling in in Qrry mode...until it is rebooted and it will work for a few more minutes. The above log is shortly after turning it on and stops as it becomes unresponsive I can provide a longer one if needed.

> What is needed is a hive mind mode where all the nodes intelligently issue query commands from closer nodes

Is that how Room-Assistant works?

DTTerastar, 2022-01-31:
Room assistant doesn't query for model number, but it does have an app that we can also use to assign a guid to a device.

Zipties, 2022-01-31:
> Room assistant doesn't query for model number, but it does have an app that we can also use to assign a guid to a device.

Yeah I have been playing with the app. It seems to randomly stop broadcasting, and I don't think it has an apple watch component that I can tell.

DTTerastar, 2022-01-31:
It definitely doesn't stay running 100%...  And yes, no apple watch.

Zipties, 2022-01-31:
Is there a way to detect any watch or phone in a room with less overhead, without caring which device it is specifically?

DTTerastar, 2022-01-31:
If you turn on the covid contact tracer apps you can get exp:20 ids and count them.

Zipties, 2022-01-31:
> If you turn on the covid contact tracer apps you can get exp:20 ids and count them.

but again, no watch right?

What if anything would need to change if multiple devices were reporting the exp:20 ID to home assistant? Wouldn't that cause flip flopping issues?

DTTerastar, 2022-01-31:
Yes, mqtt_room would be unhappy with that, but we could make something that counts the same id with different macs.  It could count up all the "kinds" of things, watches, phones, laptops, and a total count.

Zipties, 2022-01-31:
I have also noticed that even with query off, thumbprints will start randomly working for some devices, even days later. I set up my girlfriends, and my apple watches a couple of days back but they stopped working shortly after disabling query. But, they pop back up and work great for a couple of hours every day still using the apple:1005:9-24 / 9-26 thumbprints

closed 2022-04-10 by stale[bot]