ESPresense/ESPresense #215: Known issues with Apple Watch?

oskarax, 2022-02-01:
Hi, 

Im having issues with my apple watch not becoming "not_home" it goes from 4-8 meters in the distance even though the Apple Watch is turned off.

Using the default settings.

**Config**

- [ ] Active Scan enabled
- [ ] Query enabled

DTTerastar, 2022-02-09:
You may have multiple apple watches all using the same id?  Check the mac addresses and see if they keep switching back and forth.  That means you're seeing 2 or 3 devices under the same id.

PDekker, 2022-02-10:
> You may have multiple apple watches all using the same id? Check the mac addresses and see if they keep switching back and forth. That means you're seeing 2 or 3 devices under the same id.

Is there a solution to this? My wife and I both have the same Apple Watch model. Her's is smaller though.

closed 2022-04-18 by stale[bot]