ESPresense/ESPresense #216: IPhone inconsistently reporting

begunfx, 2022-02-03:
**Describe the bug**
Not sure if this is an espresense issue or just the way the iPhone works, but it seems like I'm often loosing the iPhone Bluetooth signal in my espresense network.  I'm only relying on the iPhone's built-in Bluetooth and not any App.  

**Expected behavior**
iPhone consistently reporting its status within espresense sensor network.

**Hardware Details**
iPhone 13.

**Config**

- [ ] Active Scan enabled
- [ ] Query enabled

**Logs**
If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**
If applicable, add screenshots to help explain your problem.

Stooovie, 2022-02-09:
It's essentially useless on my 11 Pro. Infrequent updates, bogus distances.

k41zen, 2022-02-10:
I have the same issue here too. Bluetility spots my iPhone pretty much straight away sitting right next to my single ESP32. But watching in MQTT Explorer the phone intermittently shows. When it does show in the list it updates quite frequently. Can't watch through the terminal session really as I'm getting the reboot issue too. Only installed today.

DTTerastar, 2022-02-11:
Which id are you using?

k41zen, 2022-02-11:
Mine is apple:iphone14-2. I got this from MQTT Explorer and also the iphone14-2 matches in Bluetity.

davidcreager, 2022-02-13:
I am just at the beginning of my espresence journey -:).  I am also noticing this problem.   I just noticed the following message in the log 

```
21:11:35.513 -> lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
21:11:35.560 -> 1 QryErr| MAC: 6794171cd43d, ID: apple:1006:10-12                                             rssi -74, try 0, retry after 3ms
21:11:36.351 -> lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
21:11:36.351 -> 1 QryErr| MAC: 6794171cd43d, ID: apple:1006:10-12 
```
I then turned off bluetooth on the phone and turned on again and the esp32 found the device and behaved properly.

Not sure if this helps or not

davidcreager, 2022-02-14:
Yes thanks....much appreciated that a good engineer is looking at this......I have wasted hours on this sort of problem with various projects. My (inexpert!) opinion is that the ESP32 ble implementation is too buggy for reliable repeatable approaches.   However, I hope that an approach that allows for this could be very successful for this use case

Good luck!!!!

begunfx, 2022-04-15:
I'm closing this as I've seen consistent correct results from my Son's IPhone for some time now.

closed 2022-04-15 by begunfx

DTTerastar, 2022-04-17:
Awesome thanks!