ESPresense/ESPresense #219: add attributes to MQTT message: trigger source and lastMoved timestamp

wivaku, 2022-02-04:
as a result of #198

> We send based on a timer as well, because mqtt_room will mark it as away if no signal received.
> So it is whichever comes first: either 1) distance moved or 2) time elapsed?
> Correct.

When processing the MQTT message I would like to know what triggered the message: did it move or did the time elapse?
And if it did not move: when was the last time it *did* move?

Request: add attributes to the MQTT message:
- `trigger: time` or `trigger: distance` to show what triggered that message
  (# 198 also suggests `trigger: offline` but I believe mqtt_room already takes care of that by marking the beacon as away)
- `lastMoved: <timestamp>` so the receiver knows when the beacon was last seen moving
  definition of `lastMoved`: when was the last time difference between distance of current and previous sample was bigger than threshold

DTTerastar, 2022-02-09:
Those are great ideas, but i'm trying to limit the amount of data we send on every send, and basically it will be nonsense because rssi always jitters and moves due to multipath reflection.  You would just get bad data that wouldn't be helpful to automate based on.

ralf-koe, 2022-02-11:
> 

Let me try to explain my view at the moment. I don't use HA, based on that I don't know what is done there. My expectation is a bit, that "this espresense" deliver the necessary information for MQTT capable systems.

If I enable "telemetry", I get every minute messages with a low amount of changed content and "only" system  monitoring data.

Within "device" I get information related to a "tracking item" I want to see. But if the beacon is not longer seen, I didn't get any suitable info from espresense within this topic, the data will "only be old" in automation/MQTT system. 

But espresense know, that the beacon is not longer visible (or not moved), why it don't provide this information?

I test with 4 ESP32 and 
Update mqtt if beacon has moved more than this distance since last report (in meters): 2 
Update mqtt if this time has elapsed since last report (in ms): 10000

My MiFloras and plant potts are not jumping around, but I still get a large ammount of "devices" messages.

closed 2022-04-19 by stale[bot]