ESPresense/ESPresense #220: using beacon connecting to isearching® app on iphone

ironjohnjohn, 2022-02-05:
Hello 

i want to us espresense for tracking some beacon, i ve got a lot of tracker that i bought on aliexpress and they are connect through isearching app on my iphone.
i read the documentation on espresense and i 'm not sure they will work .
does anyone already used them 

thx for your help

DTTerastar, 2022-02-09:
I have used them.  They should show up as "Close" on the terminal when you turn it on and have them touch the esp32.  Just note the id shown and then you can put that into HA.

ironjohnjohn, 2022-02-14:
Thx it work

closed 2022-02-14 by ironjohnjohn

ozett, 2022-03-01:
i am looking for cheap bluetooth tiles that are discovered from esp without pressing a button.
@ironjohnjohn : do you got some from ali that run on battery and could be tracked all the time by distance, without interaction and app? some links? 
(i am not sure if i can track distance of tiles with only bluetooth 4.0 that way, maybe must be bt5/LE for powersafe)