ESPresense/ESPresense #223: RSSI @ 1m doesn't seem to work

trullock, 2022-02-08:
**Describe the bug**
When I set the RSSI@1m setting via the captive portal, this doesn't seem to have any effect on what ends up in MQTT. Should it?
The captive portal seems to save and remember the value between refreshes/restarts, but its not what ends up in MQTT

**Expected behavior**
The `"rssi@1m"` value of the room under `espresence/rooms` to be whatever I set it as via the captive portal.
Currently its always `"rssi@1m": -100,`

**Hardware Details**
Its running on an ESP32 Devkit

**Config**

- [ ] Active Scan enabled
- [ ] Query enabled

^ These don't seem to make any difference

**Logs**
N/A

**Screenshots**
N/A

peetipablo, 2022-02-11:
Hello,

I can confirm that adjusting the 'rssi@1m' setting in the webui doesn't always update the actual setting.  

Steps to reproduce:

1. Change 'rssi@1m' setting in webui
2. Click 'save'
3. Click 'restart'
4. MQTT explorer doesn't show changed 'rssi@1m' value for devices with updated reports.

What is strange is that this is occurring on one of my esp32 dev boards, but for another of my esp32 dev boards I was able to change the value to -70 without issue.

trullock, 2022-04-12:
This is still broken

DTTerastar, 2022-04-12:
I need more info on why you expect it to work otherwise.   The -100 is because your using an iBeacon w/ that value in it.

DTTerastar, 2022-04-12:
iBeacons transmit a rss@1m.  It's in the settings for HA Companion to change it.  I have updated the documentation to note this a while back: https://espresense.com/configuration/settings#calibration

trullock, 2022-04-20:
Ah ok, thanks

closed 2022-04-20 by trullock