ESPresense/ESPresense #230: Putty serial not displayed correctly

Papi77, 2022-02-09:
**Describe the bug**
When connected through putty the lines in terminal are displayed shifted.

**Expected behavior**
Text in putty terminal should be formatted correctly

**Hardware Details**
ESP32-WROOM, Windows 10 PC.

**Config**

Putty configured with:
- Baud speed: 115200
- Data bits: 8
- Stop bits: 1
- Parity: none
- Flow control: XON/XOFF

**Logs**
If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**
[image]

wabit, 2022-02-09:
Try setting implicit CR in putty's settings their documentation explains it well
[puttydocs](https://documentation.help/PuTTY/config-crlf.html#:~:text=3%20'Implicit%20CR%20in%20every%20LF'&text=The%20CR%20character%20makes%20the,might%20make%20the%20screen%20scroll).)

Papi77, 2022-02-10:
Thanks, yes that seems to fix the issue with putty!

closed 2022-02-10 by Papi77