ESPresense/ESPresense #231: Love the idea but having a hard time make it work consistently!

jcastro, 2022-02-09:
**Describe the bug**
I have setup 3 Wemos D1 Mini ESP32 nodes around the house. I'm setup 2 devices to track (my iPhone and Apple Watch) however even next to the node sometimes they will jump to "not_home" state. Also sometimes it will take many many minutes to change reported location from one room to another. Not entirely sure how to make it work more reliably to me considering I'm using Apple devices

**Expected behavior**
Should track devices consistently.

**Hardware Details**
Apple iPhone 13 Pro
Apple Watch 7

**Config**
All 3 nodes are configured the same, as the image shows below
[image]

**Logs**
If possible, copy and paste any logs from the serial monitor from the time that you see the bug.

**Screenshots**
If applicable, add screenshots to help explain your problem.

the4jaggz2, 2022-02-09:
Experiencing the exact same issues as you using my Apple Watch 6 only (found that with my iPhone 13, it was even more flakey). MQTT Explorer indicates that the distance of the device (which appears to be reporting consistently/frequently) is well within the "maximum distance to the report" and subsequently, Home Assistant shows the same yet for some unknown reason, it randomly goes to the 'not_home' state.  In fact, when I go to HA developer tools and select the entity, it confirms the distance but will show 'not_home' even when I hit the refresh button.  As well, when it is working, I do find the reporting state refresh rate varies but generally it's a few minutes versus seconds...more when entering into the 'not_home' state.  Would love to understand if this is a bug or whether there are settings that need to be adjusted.  I have played with the "Forget beacon..." and "Update mqtt if beacon has moved.." settings but have found that it has not changed the stability.   Happy to help with any trouble shooting or provide logs as necessary as this seems like a really cool application.

DTTerastar, 2022-02-09:
Can both of you monitor your uptime and see if that correlates to it going not_home: 

```- platform: mqtt
  state_topic: espresense/rooms/kitchen/telemetry
  value_template: '{{ value_json.uptime }}'
  unit_of_measurement: 's'
  name: 'ESPresense Kitchen Uptime'```

jcastro, 2022-02-09:
Right now both my phone and watch are next to the bedroom node, however one if them is being identified but the other is showing not_home. The uptime it's around 44 minutes while the last change was 13 minutes
<img width="370" alt="Screenshot 2022-02-09 at 22 37 11" src="https://user-images.githubusercontent.co
[image]
m/190036/153294566-5de73717-9ac0-4b50-b2ed-c27b286634f2.png">

the4jaggz2, 2022-02-09:
Based on my first time tracking test, it appears the state randomly changes at the 44 min mark (basement to not_home in my case) which interestingly enough, coincides to what was indicated in this post of another solution (https://github.com/dalehumby/ESPHome-Apple-Watch-detection). More specifically, "Apple randomly generates a new MAC address about every 45 minutes" for privacy reasons.  While the state does correct itself after a few minutes, for obvious reasons, any automations based on the state become unreliable.  Is there anyway around this one for Apple Devices or is it simply not possible for ESPresense to accommodate?

the4jaggz2, 2022-02-09:
Appears it's more random than that as the next state change happen 19mins later...meaning it was reporting the correct state for 19mins (i.e. basement), before going into the not-home state.  There was no drop out of the ESPresence/MQTT service that would have caused it. Once again, the state corrected itself but it took 14+ minutes versus just over a minute the last time.  So all very random...

jcastro, 2022-02-10:
Yeah same here, I'm literally next to the ESP32 node for both my devices but they will randomly change to not_home or other room even if each module is more than 4 meters apart. I don't seem to be able to get this working stable enough, sadly. Not sure if I'm doing something wrong or it's just because of the Apple devices

DeekB1983, 2022-02-13:
I am also going through the same struggle with a couple of I-phones and an I-watch too, Reading the posts it seems like there are many users that have issues with tracking apple watches and phones.

One thing that has improved things a little bit more me is increasing the not_home timeout value to 10 minutes from 2 minutes.

Also I find when any of the Iphones or watches change to not__home when they are actually home the only way I can get them to report home again is by restarting the device itself.

I am also using a mi-band 6 and a couple of Android phones and have no issues at all, to me it seems more so an issue with apple devices as a whole

DeekB1983, 2022-02-13:
> > I am also going through the same struggle with a couple of I-phones and an I-watch too, Reading the posts it seems like there are many users that have issues with tracking apple watches and phones.
> > One thing that has improved things a little bit more me is increasing the not_home timeout value to 10 minutes from 2 minutes.
> > Also I find when any of the Iphones or watches change to not__home when they are actually home the only way I can get them to report home again is by restarting the device itself or waiting which can vary in time scales, sometimes its back within 5 minutes sometimes its 30 minutes, there is no real pattern to it.
> > I am also using a mi-band 6 and a couple of Android phones and have no issues at all, to me it seems more so an issue with apple devices as a whole
> 
> I’m currently testing https://github.com/dalehumby/ESPHome-Apple-Watch-detection and it seems to work really well, plus it runs with ESPHome so I don’t have to add more ESP32 modules!

That does look like it is worth a try but guessing I would need more modules running that as well as ESPresence as I am tracking more devices than just apple watches in my house hold.

closed 2022-04-22 by stale[bot]