ESPresense/ESPresense #234: iPhone stops reporting randomly

nickelnine, 2022-02-10:
**Describe the bug**
iPhone 12 Pro went not_home all of a sudden and has not gone back to bedroom like it was doing earlier today. It is currently 10PM MST and the last reported message in MQTT Explorer is from 9:47PM and the phone is a little more than a meter from the ESP32. I launched room-assistant app and immediately saw MQTT messages pouring in but when I configured mqtt_room in HA and rebooted, the MQTT messages stopped arriving for room-assistant. Both methods are frozen and nothing in MQTT is occurring for these 2 but my Android phone is chugging along just fine. I rebooted the ESP32 but no dice.

**Expected behavior**
MQTT messages to keep coming and not suddenly stop.

**Hardware Details**
iPhone 12 Pro latest iOS 15.x and running room-assistant app

**Config**

- [ ] Active Scan enabled
- [ ] Query enabled

**Logs**
NA
**Screenshots**
NA

leacho73, 2022-02-10:
I can confirm that the iPhone 13 also suffers from this random issue - I left my iPhone next to the ESP all day yesterday and today - and get the following results:
[image]

jcastro, 2022-02-10:
Same here! using iPhone 13 Pro + ESP32 Wemos D1 Mini

DTTerastar, 2022-02-11:
What are the ids you're using?

k41zen, 2022-02-11:
I'm using apple:iphone14-2 which is what I do sometimes see in MQTT Explorer. Bluetility shows it too:
[image]

nickelnine, 2022-02-11:
Initially "apple:13-3" but after a reboot it appears as "apple:1005:9-12".

tobst96, 2022-02-12:
Same here. Apple don't work. 
iPhone XR
iPhone 11
2 x Apple Watch Series 7

DanielBaulig, 2022-02-13:
I am having trouble with spotty reporting for an iPhone 14 Pro, iPhone XS and an Apple Watch Series 7, too.

DTTerastar, 2022-02-14:
The newest version allows you to selectively decide which ids you want queried to improve the id.  You can now put "apple:1005:9-12" in the query box and it'll ask the phone for it's model and turn it into "apple:13-3".  BUT if apple:1005:9-12 is working for you reliably it's better just to use that and keep query empty.  The ESP32 has to stop listening while sending out the queries which hurts the reliability of receiving advertisements from devices.

abreups, 2022-03-02:
Same issue with iPhone X, iOS 15.3.1 and ESPresense v2.0.34.

nickelnine, 2022-03-23:
Looking at this more closely again, the model for my iPhone - apple:iphone:13-3 is sometimes listed in MQTT and hence successfully reporting and sometimes not - thus the randomness. I have the device ids in the Query field but can't nail down a pattern to the missing model. These are the device ids I set - apple:1005:9-12 apple:1005:9-8 apple:1005:9-26

afinzel, 2022-05-08:
Just wanted to add to this.  It looks like the device fingerprinting doesn't always work.  I have set up room assistant on the devices.  It will work but then I think the mac changes and the device is no longer registering as roomassistant:xxxx-xxx but it has gone back to apple:1006:10-8.  I saw this behaviour by using MQTT explorer and seeing the mac coming up under multiple devices depending on the esp.  Is it easy to debug this code?

DTTerastar, 2022-07-18:
The newest beta has a feature to "Enroll" iOS devices.  We store the BLE IRK and from that point on you can use that irk:{32chars} to uniquely identify that iOS device!

afinzel, 2022-07-18:
That is great news.  Thanks

leacho73, 2022-07-20:
> The newest beta has a feature to "Enroll" iOS devices. We store the BLE IRK and from that point on you can use that irk:{32chars} to uniquely identify that iOS device!

This is great, thanks, when will this Beta be available?

DTTerastar, 2022-07-20:
Check the box labeled "Prerelease" and you'll have it in 5 minutes!

leacho73, 2022-07-21:
Ah cheers @DTTerastar - I am running 2.2.8 - but couldn't see how to interview the apple devices, so assumed I didn't have it :)

DTTerastar, 2022-09-01:
It's released, going to close this.

closed 2022-09-01 by DTTerastar