ESPresense/ESPresense #239: ATC MiThermometer add battery percentage value.

TheDukeSr, 2022-02-10:
**Is your feature request related to a problem? Please describe.**
Only battery mV is now used, I'm missing the "battery %" in MQTT

**Describe the solution you'd like**
Add the "**battery %**" value to MQTT

**Describe alternatives you've considered**
Calculating % in Hass from the mV, but that's realy complicated and extra work for Hass PC then just getting it from the Thermo itself.

**Additional context**
I do get battery % when using the ESPhome LBE code, so it's proven available to use.
I've found this link where ATC was added to the code in previous version:
[https://github.com/ESPresense/ESPresense/pull/124#issue-1050517543](https://github.com/ESPresense/ESPresense/pull/124#issue-1050517543) 
I hope the Battery Percentage can be included.

Regards,

The Duke.

DTTerastar, 2022-03-11:
I did this in the prerelease, please test.

TheDukeSr, 2022-03-13:
> I did this in the prerelease, please test.

Sorry, but i have no idea how to install/test a prerelease version.
Do you have a step by step guide?

DTTerastar, 2022-03-22:
All you have to do is check the box.  It's released now.  Closing...

closed 2022-03-22 by DTTerastar