ESPresense/ESPresense #242: Version: v2.0.17 Temperature sensors gone

rene-bayer, 2022-02-11:
**Describe the bug**
Since the latest update it's not possible to configure temperature sensors anymore

**Expected behavior**
configure temperature sensors (DHT11, DHT22)

**Hardware Details**
Lolin32 & DHT22

**Config**

- [x] Active Scan enabled
- [x] Query enabled

**Logs**
NaN

**Screenshots**
[image]
[image]

Novocaine85, 2022-02-11:
Same for me. Got 3 around my house and none is working anymore.

rene-bayer, 2022-02-11:
@Novocaine85 Don't know why, but there's a separate version to use with sensors now: [https://github.com/ESPresense/ESPresense/releases](https://github.com/ESPresense/ESPresense/releases)

closed 2022-02-11 by rene-bayer