ESPresense/ESPresense #244: "Motion" not detected

dvbit, 2022-02-11:
I have configured ESPresense to get the input from pin G22 of an atom lite.
ON this pin I have connected a pressure mat : https://uk.farnell.com/defender-security/pm3-pk/pressure-mat-large-720x560mm/dp/1146764 on one pin the other wire of the mat on the ground pin of the atom lite

I hoped, after reading the code, I could use this to detect the presence of a person on the bed (espresense will be under it :-))

I have not found in the docs where I should read the motion sense in the Matt but I suppose it is in the "motion" topic.
Unfortunately there I keep getting a toggling of ON / OFF when the mat is connected; a steady OFF when the mat is not connected and static ON/OFF when moving the atom lite with my hand

Fully aware this is not a PIR so I understand it could not work (I will try to modify the source to handle it in this case) but can you please confirm or rule out the motion PIN could work in that configuration?

Thanks in advance

- [X ] Active Scan enabled
- [ X] Query enabled

DTTerastar, 2022-02-21:
I think the pin should work fine like this.  If you made a very simple Arduino sketch you could prove it out before adding ESPresense to the mix.

dvbit, 2022-02-24:
Thanks. Meanwhile I switched to another sensor type st15-600.
Resistive strip . Testing with esphome resulted in the need to set the pin as pull-up. Can this be also parametrises in the current espresense code?
Ti

closed 2022-05-28 by stale[bot]