ESPresense/ESPresense #251: Home Assistant companion app iBeacon isn't detected

erbmur, 2022-02-12:
I have a couple esp32's set up around the house.  They are picked up a while bunch of apple devices (which I don't own), but it doesn't appear to be detecting either mine or my wifes android phone.  I have the HA companion app installed and the iBeacon from there isn't being detected either.
I installed beacon simulator and set up an ibeacon on that which is being picked up by the esp's.
Is there something specific I have to do to have the beacon from the HA companion app detected?  I know it's working with a separate app, but I would rather not have extra apps installed on the phones if I can help it.

DeekB1983, 2022-02-12:
Have you set the power to high?

erbmur, 2022-02-12:
I was unable to find any options within the companion app to turn BLE on/off or anything to do with power levels.
I found something on the home assistant website that allows turning it on and off via a notify command.
Is there somewhere in the HA companion app settings you can point me to that would allow me to make the adjustments?

erbmur, 2022-02-12:
LIFESAVER!

closed 2022-02-12 by erbmur

bachoo786, 2022-04-01:
> Go into the companion app, go to manage sensors, then Bluetooth sensors, then into Bluetooth transmitter, enable the setting and then scroll to sensor settings at the bottom.
> 
> You will find 2 settings, advertise mode and transmitter power.
> 
> Set transmitter power to high Set advertise mode to 10Hz low latency
> 
> These settings are working well for me on my Samsung galaxy z fold 3
> 
> [image]

oh I wish this worked for me I have been struggling with my HA companion app on my Samsung Galaxy S20+ and my entity in HA says unavailable. 

I tried your setting and I still have the same issue.

DTTerastar, 2022-04-01:
The only other thing is sometimes HA companion broadcasts with such a low power max distance kicks in and hides it.  Set max distance to 0 to be sure it's not being excluded.

bachoo786, 2022-04-01:
> I'd bet you have them under iBeacon:*
> 
> Can you show what MQTT explorer sees?

sorry how do I show what the MQTT explorer sees?

bachoo786, 2022-04-01:
> Contributor

ok 

> I'd bet you have them under iBeacon:*
> 
> Can you show what MQTT explorer sees?

ok just checked my MQTT explorer app and you are correct it is seen there, please see screenshot below:

[image]

DTTerastar, 2022-04-01:
So you just need to configure the id as the full iBeacon:exxxx100-1 in home assistant and it'll work.  The documentation is a little out of date on this.  It shows it without the type: prefix.  You gotta use the whole id.

bachoo786, 2022-04-01:
> 

OMG Thank you ever so much it bloody works !!!

Amazing Thank you !!!

I have another question how can I make the system to detect my phone quickly? currently there is a delay of 5 to 10 seconds.

DTTerastar, 2022-04-02:
On your screenshot there is one more value to the right.  Interval.  What is it?  That's how often we're seeing beacons in ms.

bachoo786, 2022-04-03:
> On your screenshot there is one more value to the right. Interval. What is it? That's how often we're seeing beacons in ms.

its the interval value and it ranges from 4800 to 5100

bachoo786, 2022-04-04:
Is there a way to increase from 5 seconds to 1 second?

Also I have set up 4 espresense and the issue I am facing is that when I am in a room say office which has 1 espresense installed, it sometimes says I am in the kitchen where there is another espresense installed. 

How can I resolve this and make it accurate?

Thanks again :)

DTTerastar, 2022-04-04:
The update interval is a setting in home assistant companion app.

bachoo786, 2022-04-04:
> The update interval is a setting in home assistant companion app.

Ok thanks for that. How about the distance? 

Basically I have my base station in my office and the rooms next door do not have a base station. Whenever I walk into the rooms next to the office, the signal is picked up as I am in the office but clearly I am not. Is there a way where I could make sure it detects me when I am in the office only and not near the office?

DTTerastar, 2022-04-04:
You can tweak the maxDistance to decrease the radius in which the base station will report.  You can reposition the base station to be further away from the area you don't want to report in.  Ultimately I want to run all the distances through some machine learning to make it more precise but baby steps first...

bachoo786, 2022-04-04:
> You can tweak the maxDistance to decrease the radius in which the base station will report. You can reposition the base station to be further away from the area you don't want to report in. [...]

That would be amazing 👏 
I remember you mentioned to keep the maxDistance as 0 when the power is set to high as the HA app doesn't transmit it correctly ?

DTTerastar, 2022-04-04:
Sure if you're NOT seeing devices set it to 0.  But here you're see too many devices, so set it to a value that works for your house.

bachoo786, 2022-04-04:
> Sure if you're NOT seeing devices set it to 0. But here you're see too many devices, so set it to a value that works for your house.

Ok great makes sense. How about the factor for absorption? Default is 3.5. I have seen when I play with the value and decrease it my results change drastically.

DTTerastar, 2022-04-04:
https://espresense.com/configuration/settings#calibration

> Factor used to account for absorption, reflection, or diffraction - An average value that accounts for the amount of “stuff” in between rooms in your house. Use a higher number for dense materials, lower for assuming no walls.

bachoo786, 2022-04-06:
> https://espresense.com/configuration/settings#calibration
> 
> > Factor used to account for absorption, reflection, or diffraction - An average value that accounts for the amount of “stuff” in between rooms in your house. Use a higher number for dense materials, lower for assuming no walls.

Well ESPresense is amazing but to be very honest with you it's very difficult to get the distances correct for each base station I tried it many times with different settings using the ble transmitter from the HA Companion app.

I wish there was a simpler way to make the distances defined on the settings page to just work and not show I am in the room on the floor below whilst in reality I am on the 1st floor or 2nd floor.