ESPresense/ESPresense #252: Reporting fitbit

carlywarly, 2022-02-12:
Within our family we have 4 fitbits.
Sense
Charge 3
2 x Versa 2

The ID for both the Sense and Charge is sd:0x180a and the both Versa's report sd:0x180a21.  So I cannot tell the difference between them.

Is it possible to add the mac address to the ID as default?  So the Sense id would be  sd:0x180a:d23fdxxxxxxx and the Charge sd:0x180a:cd080xxxxxxx and the Versa 's sd:0x180a21:d2cadxxxxxxx & sd:0x180a21:d3dfbxxxxxxx.  This way I can separate them in HomeAssistant.

Many thanks,
Carl

Major-Trip-Hazard, 2022-02-13:
Hello,

I have a similar situation with multiple Fitbit Charge 2 devices reporting an id of sd:0x180a-5.
We also have multiple Garmin devices that report an id of md:0087:4.
So it would be useful to have the option to add the mac address to the id.

Thanks
Paul

tonyyng, 2022-02-13:
Charge 4 also reports sd:0x180a.  I tried turning on "Query devices for characteristics (helps apple fingerprints uniqueness)", but the ID didn't change.

Thanks

tonyyng, 2022-02-14:
I posted a workaround in this issue: https://github.com/ESPresense/ESPresense/issues/260

carlywarly, 2022-02-14:
@tonyyng, working perfectly,  thank you.

tasi333, 2022-02-15:
Yes I agree, it would make it easier to use MAC for ID, especially with Fitbit devices.

MichaelStruck, 2022-02-20:
While there was an update to x.x.31, the fitbit issue remains. However, many devices now seem to have better MAC address indentifiers, so hopefully this will come to the fitbit soon.

MichaelStruck, 2022-02-25:
2.0.34 still has the issue with Fitbits having one fingerprint across multiple devices...Is there an ETA on this being fixed?

MichaelStruck, 2022-03-07:
Just tested today with 2.0.36 and I am now getting a differentiation between the Fitbit devices. This still needs work....my Fitbit sense 'fingerprint' is name:sense but what if there were two Senses on two different people? Fortunately, in my case, my wife has a Charge 4 so here fingerprint is "name:charge-4".

fmarkf, 2022-03-07:
I'm using Notifications feature on my Fitbit Versa 2 (which I guess is keeping my phone always connected to the watch). 
I don't see my Versa 2 in ESPresense(v2.0.36) terminal unless I disable Bluetooth on my phone. 

Are you guys able to track your Fitbit using ESPresense **with** Notifications feature enabled (when phone notifications are displayed on the watch)? 

Thanks.

ihrapsa, 2022-03-07:
Nope. I think fitbits stop advertsing once they get paired. I think that's how trackers work. They stop advertising once they establish a connection with a device.

MichaelStruck, 2022-03-08:
I am seeing the same thing...when the cell phone and watch are in the same room (i.e. they are connected) I don't see anything in the room. When I leave the room and go into another room it senses...

DTTerastar, 2022-03-11:
Should we put a warning in the documentation about Fitbits only working if you don't use the app?

MichaelStruck, 2022-03-13:
I would say this needs to be documented in a couple ways. First, that fact that (current) only one style of Fitbit is recognized in the household (Because of the generic fingerprint). Second, that when a Fitbit pairs with the phone, the number of BLE beacons from the watch decreases (it appears).

MichaelStruck, 2022-03-15:
Did something change again with the Fitbits? My device is now showing a fingerprint of sd:0x180a again. Would it be possible to simply substitute the MAC for this designation?

carlywarly, 2022-03-15:
No, mine are still showing as name:<whatever>, still have the problem that both of our charge 3s are showing as the same name ☹

Carl

MichaelStruck, 2022-03-22:
This issue is resolved for me using the 'known device' entry in the ESPs portal page. Device shows up as known:xxxxxx  where the xxxxxx is he MAC address. Nice solution.

DTTerastar, 2022-03-22:
Fixed documentation: https://espresense.com/beacons Github isn't building right now but once it does you can see it.

closed 2022-03-22 by DTTerastar