ESPresense/ESPresense #257: Merge apple:1006:xx-x devices with existing apple devices

calvinbaart, 2022-02-13:
**Is your feature request related to a problem? Please describe.**
I currently have an iPhone 11. This is correctly detected as apple:iphone12-1, but it sometimes takes a long time. When monitoring with MQTT Explorer I see an apple:1006:10-8 entry that has the same mac address as the apple:iphone12-1 and is constantly updated.

**Describe the solution you'd like**
If an apple device is incorrectly detected as apple:1006:xx-x but it has the same mac address as an already detected apple device (in my case both apple:iphone12-1 and apple:1006:10-8 have 6182d2c0edff as mac) it should be merged back into the already detected apple device instead of staying it's own entry. This would solve some reliability issues with apple tracking.

**Describe alternatives you've considered**
Just waiting, eventually it will be recognized as apple:iphone12-1 but that takes between 5 - 10 minutes sometimes.

**Additional context**
[image]

(Redacted room names for Privacy)

DeekB1983, 2022-02-14:
I too seem to be seeing the same issue
[image]

I see a lot of times where the 2 iPhones I am trying to track will send messages via mqtt, then they messages will stop for a long period, I am using the room assistant app on the 2 iPhones but it feels like the iPhones are just not reliable enough at consistently sending or receiving the advertisements, it would be nice if the home assistant IOS companion app had the same BLE Transmitter functionality that the Andriod companion app has as that seems very realible on the android phone I am tracking

DeekB1983, 2022-02-14:
For anyone with an Iphone having the same issue I have requested the BLE Transmitter functionality to be added to the IOS HA Companion app on this discussion thread - https://github.com/home-assistant/iOS/discussions/2081

calvinbaart, 2022-02-14:
I have made some test changes locally on my devices where if an apple device is detected the espresense sensor will broadcast on the /espresense/mac mqtt channel the id and mac of that iphone and other espresense sensors will register this as a fingerprint if it was unknown. This seems to merge them ok, but I'm still having issues getting them to transmit reliably. I think adding them to the home assistant app will probably work better.

Example of the /espresense/mac channel:
[image]

calvinbaart, 2022-02-14:
Examples with transmission reliability:
- My iPad is detected in the bedroom and living_room even though it's in the kitchen (and the kitchen has an espresense sensor)
- My iPhone 11 is only detected in the bedroom even when moving through the house (I have 5 rooms setup)
- Another iPhone 13 is only detected in the living_room and kitchen even though it's currently in the bedroom.

When I reboot the espresense devices they will detect the devices correctly, but then other rooms stop detecting them.

closed 2022-04-23 by stale[bot]