ESPresense/ESPresense #260: Multiple fitibit devices with different BLE MAC ID's all show the same ID for espresense

AbhiOommen, 2022-02-13:
**Describe the bug**
Two separate Fitbit devices (ACE 3), child devices linked to the same family/parent account, both  with different Bluetooth MAC addresses show the same ID and hence can't be tracked separately.

**Config**

- Version: v2.0.20
- Active scan enabled: Yes
- Include filter: 
- Exclude filter:
- Query filter:

**Hardware**

- Device (be specific): Fitibit Ace 3 Pedometers/Trackers

tonyyng, 2022-02-14:
I have a workaround for this situation which seems to do the trick for me.  In HA, I register my Fitbits using their mac instead of the "sd:0x180a" id that ESPresense is sending.

Then, I use a Node-Red flow to subscribe to messages with topics espresense/devices/# and espresense/rooms/#.  For any messages that have the id that matches my Fitbits, the flow republishes the message using the mac as the id.

[image]

Here is my flow,  You'll need to modify the function node and change "sd:0x180a" to the id that ESPresense is reporting for your devices.

[image]

`[{"id":"0f4b1e750f6d4649","type":"tab","label":"ESPresense","disabled":false,"info":"","env":[]},{"id":"e448d63a6c7ec391","type":"mqtt in","z":"0f4b1e750f6d4649","name":"espresense/devices","topic":"espresense/devices/#","qos":"2","datatype":"auto","broker":"97cecd81c46e5aba","nl":false,"rap":true,"rh":0,"inputs":0,"x":110,"y":120,"wires":[["f5da739d51836379"]]},{"id":"8544147d70af0dfd","type":"function","z":"0f4b1e750f6d4649","name":"Set ID = MAC","func":"if (msg.payload.id == \"sd:0x180a\"){\n    msg.topic = msg.topic.replace(msg.payload.id, msg.payload.mac);\n    msg.payload.id = msg.payload.mac;\n    return msg;\n}\nreturn null;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":480,"y":120,"wires":[["7f4aa4e306c82863"]]},{"id":"7f4aa4e306c82863","type":"mqtt out","z":"0f4b1e750f6d4649","name":"publish adjusted message","topic":"","qos":"","retain":"","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"97cecd81c46e5aba","x":710,"y":120,"wires":[]},{"id":"445a66982c07eaa9","type":"mqtt in","z":"0f4b1e750f6d4649","name":"espresense/rooms","topic":"espresense/rooms/#","qos":"0","datatype":"auto","broker":"97cecd81c46e5aba","nl":false,"rap":true,"rh":0,"inputs":0,"x":110,"y":200,"wires":[["f5da739d51836379"]]},{"id":"f5da739d51836379","type":"json","z":"0f4b1e750f6d4649","name":"","property":"payload","action":"","pretty":false,"x":310,"y":120,"wires":[["8544147d70af0dfd"]]},{"id":"97cecd81c46e5aba","type":"mqtt-broker","name":"","broker":"mqttserver.local","port":"1883","clientid":"","autoConnect":true,"usetls":false,"protocolVersion":"4","keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","birthMsg":{},"closeTopic":"","closeQos":"0","closePayload":"","closeMsg":{},"willTopic":"","willQos":"0","willPayload":"","willMsg":{},"sessionExpiry":""}]`

For a general solution, what about these options:

- An option to send two messages per device:  one with the BLE Fingerprint and a second message with the MAC
- An option to specify a list of devices that the MAC should be used as the identifier.  This option would be specified with a list of macs.

carlywarly, 2022-02-16:
I change the code a little, as both laptops were reporting the same ID.

[image]

if(msg.payload.id) {
    var id = msg.payload.id;
    if (id.indexOf(msg.payload.mac) == -1){
        msg.topic = msg.topic.replace(msg.payload.id, msg.payload.id + ":" + msg.payload.mac);
        msg.payload.id = msg.payload.id + ":" + msg.payload.mac;
        return msg;
    }
}
return null;

This way, any ID that doesn't have the MAC address in it, gets added:

[image]

Major-Trip-Hazard, 2022-02-16:
Thanks @tonyyng & @carlywarly for the workaround etc, installed Node-Red and working perfectly.

tasi333, 2022-02-16:
Thanks @tonyyng @carlywarly amazing work.

@DTTerastar any chance we can get a firmware fix in the next release?  Would love to use our fitbits for tracking natively.

MichaelStruck, 2022-02-19:
+1 for this issue...exact same issue as AbhiOommen...two fitbits, two different MACs, but the same fingerprint (sd:0x180a). Not sure how the fingerprint is calculated, but without the NodeRed solution above this makes it impossible to trace two separate trackers. Thanks!

ozett, 2022-02-20:
yeah, after some fiddling i also think that the fingerprinting should improve to identify all devices.
hard to tell which is which for my 10+ apple devices at home

MichaelStruck, 2022-02-20:
The firmware for the ESPresense was bumped to x.x.31 just now, but the issue with the Fitbit remains :(

ozett, 2022-02-21:
are bluetooth mac-adresses similar to ethernet mac-adresses?

can only find this vendor-mac for fitbit in the source,
https://github.com/ESPresense/ESPresense/blob/4780e362ae4a376148ca2bbe2fb8c72b4b398459/lib/BleFingerprint/util.h#L15

and fitbit in this mac-adress list for ethernet.
https://gist.github.com/aallan/b4bb86db86079509e6159810ae9bd3e4#file-mac-vendor-txt-L13740

should the ethernet-mac go also into the fingerprinting code, 
to identify this devices ? or something more wrong with the fingerprinting?

Major-Trip-Hazard, 2022-02-21:
I don't think it is just fitbit devices that are affected as I  also have multiple Garmin devices that report an id of md:0087:4.
Thanks
Paul

ozett, 2022-02-21:
the manufacterer for a tile seems to be defined as BLEUUID, but not build into a fingerprint like the other

https://github.com/ESPresense/ESPresense/blob/4780e362ae4a376148ca2bbe2fb8c72b4b398459/lib/BleFingerprint/BleFingerprint.cpp#L258

it should be possible to have more devices form one manufacturer, 
only the same one short ID for all the same devices seems not to be enough.
but as i am no coder, to solve this part is for someone else.

MichaelStruck, 2022-02-25:
2.0.34 still has the issue with Fitbits having one fingerprint across multiple devices...Is there an ETA on this being fixed?

ihrapsa, 2022-03-07:
I'm having the same issue with 2 fitbits. We've both got charge 3. I  was able to differentiate between them based on their mac a few couple of versions ago but not anymore (not sure what happened as their ids have changed from mac to `name:charge-3`)

DTTerastar, 2022-03-10:
The known_macs feature in prerelease should allow you to turn any mac into a unique id.  Will this resolve your issue?

MichaelStruck, 2022-03-18:
Was this known_macs feature get published? For some reason my fingerprints for the couple Fitbits we have in our house reverted from the model name back to the sd:0x180a

MichaelStruck, 2022-03-18:
I tried the known_mac feature and had to play with it to understand, but very clever! For those reading this, you place your MAC into the portal page, and the fingerprint ends up being "known:xxxxxxxx". Testing to ensure this works, but will be nice if it does!

closed 2022-03-22 by DTTerastar