ESPresense/ESPresense #262: 2 ESps in the same large room

jsBlackBelt, 2022-02-14:
**Describe the bug**
In a big room, one ESP might not be enough.
So, what will happen if I setup 2 ESPs with the same room name and install them in 2 corners of the room?

DeekB1983, 2022-02-15:
I think these ESP's can be picked up from quite a distance, I stay in the UK in a 3 bed semidetached house, I have 3 of these and find at times my trackers jump from one to the other even though they are spaced well far apart.

I dont see an issue with you have 2 in the same room but you need to ask yourself if it is really required.

I dont think you can have 2 with the same names though as I am sure when added to HA they will be added as different entities, like say your room name is Test-Room, you would end up with test_room_1 and test_room_2 entities and thats if the ESP Boards and mqtt can hand 2 with the same name.

If you are sure you require 2 in the same room them what you could do is have 1 at one of the room named Test-Room-1, at the other end another called Test-Room-2. You could then build a custom template sensor in yaml to combine the 2 rooms together as one and add any devices that you would require to occupy the sensor, Something like below:

```
  - sensor:    
    - name: "Test Room"
      state: >-
        {% if states("sensor.Test-Device-1") == 'Test-Room-1' %}
          Occupied
        {% elif states("sensor.Test-Device-1") == 'Test-Room-2' %}
          Occupied
        {% elif states("sensor.Test-Device-2") == 'Test-Room-1' %}
          Occupied
        {% elif states("sensor.Test-Device-2") == 'Test-Room-2' %}
          Occupied
        {% else %}
          Empty
        {% endif %}
```

But I guess this would only be for a really really large room, just an idea that may be worth trying to see if it fits your use case

jsBlackBelt, 2022-02-15:
well, since the home assistant sensor is based on `'espresense/rooms'`:
```
sensor:
# One entry for each beacon you want to track
  - platform: mqtt_room
    device_id: "apple:1007:11-12"
    name: 'Darrell Watch'
    state_topic: 'espresense/rooms'
    timeout: 60
    away_timeout: 120 # number of seconds after which the enitity will get status not_home
```

I guess it makes sense that 2 ESPresence NodeMCUs with the same name will publish to the same topic, so for  both NodeMCUs with the same name `someroom` - both will publish to `espresence/rooms/someroom` and by extension, if my device is close to ESP1, then it will publish and the same goes for ESP2 so regardless which ESP reports to the topic, it will recieve the message and then the HA sensor will be updated. 

I think 😅

DeekB1983, 2022-02-15:
> well, since the home assistant sensor is based on `'espresense/rooms'`:
>
> ```
> sensor:
> # One entry for each beacon you want to track
>   - platform: mqtt_room
>     device_id: "apple:1007:11-12"
>     name: [...]

Should give it a try and see, I am not 100% sure but just trying to provide you another solution you could potentially use.

I dont have any rooms large enough in my home to use 2 ESP's so have them all in separate rooms, if it was me needing 2 I woudl do it like above although your solution sound a bit more elegant if it works!

DeekB1983, 2022-02-15:
I am using these https://www.amazon.co.uk/gp/product/B08ZMLBW2R/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

I have my beacon set to High power and low latency via the home assistant companion app.

Each of my ESP32's a have the maximum distance to report set to around 3 - 3.5 metres.

closed 2022-04-25 by stale[bot]

gith-account, 2022-11-28:
See issue #600.

[DTTerastar](https://github.com/DTTerastar) (Maintainer) [on Aug 22](https://github.com/ESPresense/ESPresense/discussions/600#discussioncomment-3450043):
> Room names are required to be unique because we use it for our mqtt client id.