ESPresense/ESPresense #263: QryErr Mi Band 6

ale1800, 2022-02-14:
Hi, I have a lot of problems with Mi Band 6.
It's really slow to update (minutes) and from logs there is a Query error for every Query,

[image]

Also the distance is wrong. I can be at 4m and it shows 1m or if I'm at some centimeters it often shows a distance above 1m.

I'm using a Mi Band 6 with nfc and an ESP32 wroom 32

This is my sensor configuration:

[image]

What I'm doing wrong? Thank you in advance.

- Version:  v2.0.21
- Active scan enabled:  both
- Include filter: nothing and also tried with mifit:c10815011328
- Exclude filter: nothing
- Query filter: nothing and also tried with mifit:c10815011328

tobst96, 2022-02-14:
Same with 2.0.22
```
 Query | MAC: 57333554489d, ID: apple:1005:9-26                                              rssi -73
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
1 QryErr| MAC: 57333554489d, ID: apple:1005:9-26                                              rssi -73, try 1, retry after 5000ms
1 Query | MAC: 3bc41d5ae6aa, ID: exp:20                                                       rssi -65
1 QryErr| MAC: 3bc41d5ae6aa, ID: exp:20                                                       rssi -65, try 1, retry after 5000ms
1 Query | MAC: 227154273171, ID: msft:cdp:0902                                                rssi -80
1 QryErr| MAC: 227154273171, ID: msft:cdp:0902                                                rssi -80, try 2, retry after 6000ms
1 Query | MAC: 3a31db0419de, ID: apple:0906:10                                                rssi -86
1 QryErr| MAC: 3a31db0419de, ID: apple:0906:10                                                rssi -86, try 1, retry after 5000ms
1 Query | MAC: 3a31db0419de, ID: apple:0906:10                                                rssi -86
1 QryErr| MAC: 3a31db0419de, ID: apple:0906:10                                                rssi -86, try 2, retry after 6000ms
1 Query | MAC: 64d2c49e70d4, ID: apple:1005:9-12 
```

DTTerastar, 2022-02-14:
2.0.24 should fix the QryErrs as long as the RSSI is < -80.  Sometimes if too far away the query won't work.  But the base station that is closer should still perform the query successfully.

ale1800, 2022-02-14:
> 2.0.24 should fix the QryErrs as long as the RSSI is < -80.  Sometimes if too far away the query won't work.  But the base station that is closer should still perform the query successfully. 

Great. And are errors the reason for slow updates?

DTTerastar, 2022-02-15:
Slow updates are usually a function of how often the device is advertising.  Different devices advertise faster or slower.  For a better idea you can run the esp32-verbose flavor and watch the terminal, every packet we get is logged so you can see what we're working with.  Also try turning off your phone and see if that makes the MiBand broadcast more often.  I unfortunately haven't ever used a MiBand

DeekB1983, 2022-02-17:
What issues are you facing with the Miband?

I see similar with my MiBand6 but I am tracking it ok, the distance for my mi band also seems off at times, this is more so the device as my other devices (Mostly andriod phones and apply phones all report what looks to be a sensible distance compared to the mi band

I'm running V2.0.25 and I also see the query errors
[image]

but it does not seem to effect the tracking, I.E. I can walk from one room to another and it updates ok and I dont see it going to not home, My config has an away timeout of 300 seconds too.

yuri-vashchenko, 2022-02-19:
My Mi Band devices are not detected, none of them.
If I add `mifit:` to `Query device ids for characteristics` field it starts querying my Mi Band devices but every time with Error:

```
0 New   | MAC: 55527bdcd287, ID: apple:1006:10-12                                             031ab920011b
1 Query | MAC: f2542ae6f8bd, ID: mifit:f2542ae6f8bd                                           rssi -50
1 QryErr| MAC: f2542ae6f8bd, ID: mifit:f2542ae6f8bd                                           rssi -50, try 11, retry after 30000ms
0 New   | MAC: 7e8175489623, ID: apple:1005:9                                                 04187cadf8
1 Query | MAC: d0f6b95351c7, ID: mifit:d0f6b95351c7                                           rssi -83
1 QryErr| MAC: d0f6b95351c7, ID: mifit:d0f6b95351c7                                           rssi -83, try 13, retry after 30000ms
1 Query | MAC: f2542ae6f8bd, ID: mifit:f2542ae6f8bd                                           rssi -54
1 QryErr| MAC: f2542ae6f8bd, ID: mifit:f2542ae6f8bd                                           rssi -54, try 12, retry after 30000ms
1 Del   | MAC: 6cd03fd6bda8, ID: apple:1005:9                                                 0e18c7ad31
1 Query | MAC: d0f6b95351c7, ID: mifit:d0f6b95351c7                                           rssi -85
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
1 QryErr| MAC: d0f6b95351c7, ID: mifit:d0f6b95351c7                                           rssi -85, try 14, retry after 30000ms
1 Del   | MAC: 7e427376d708, ID: apple:1006:10-8                                              211e97909468
1 Query | MAC: f68545ddbad0, ID: mifit:f68545ddbad0                                           rssi -81
1 QryErr| MAC: f68545ddbad0, ID: mifit:f68545ddbad0                                           rssi -81, try 3, retry after 10000ms
1 Query | MAC: f2542ae6f8bd, ID: mifit:f2542ae6f8bd                                           rssi -54
1 QryErr| MAC: f2542ae6f8bd, ID: mifit:f2542ae6f8bd                                           rssi -54, try 13, retry after 30000ms
1 Query | MAC: f68545ddbad0, ID: mifit:f68545ddbad0                                           rssi -87
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
1 QryErr| MAC: f68545ddbad0, ID: mifit:f68545ddbad0                                           rssi -87, try 4, retry after 19000ms
1 Query | MAC: c668ef8ec6d3, ID: mifit:c668ef8ec6d3                                           rssi -70
1 QryErr| MAC: c668ef8ec6d3, ID: mifit:c668ef8ec6d3                                           rssi -70, try 6, retry after 30000ms
1 Query | MAC: f2542ae6f8bd, ID: mifit:f2542ae6f8bd                                           rssi -55
1 QryErr| MAC: f2542ae6f8bd, ID: mifit:f2542ae6f8bd                                           rssi -55, try 14, retry after 30000ms

```

alexreddy78, 2022-02-19:
Same problem with Mi Band fitness , after update to 2.0.28, esp32 is not detecting it at all. Downgrade to 2.0.20 solved the problem. However 2.0.28 is more stable, uptime reach 2 days without any reboot, but on 2.0.20 it's rebooting often. Would you be so kind and check possibility to track back Mi Band fitness on 2.0.28.

Regards

Alex

Jos-77, 2022-02-19:
tried downgrading, but there keeps an isue after installing. continues fail reed.
I've tried al versions whit tasmotizer an web.esphome.io as flasher. same problem  over and over.

Any advice?

DTTerastar, 2022-03-10:
As far as I know the only devices so far that support querying is apple devices.  I wouldn't waste airtime trying to query mifits especially since they use fixed mac addresses as far as I know.  Can this be closed or are you still having issues on the latest firmware?

yuri-vashchenko, 2022-03-11:
> As far as I know the only devices so far that support querying is apple devices. I wouldn't waste airtime trying to query mifits especially since they use fixed mac addresses as far as I know. Can this be closed or are you still having issues on the latest firmware?

Yes, it works fine on latest release

Jos-77, 2022-03-11:
Query turned off. Solved issue.

ale1800, 2022-03-11:
> Yes, it works fine on latest release

Can you explain how you got it working? I'm trying latest release but it's the same, only without qryerr in the log

DTTerastar, 2022-03-11:
Try turning off your phone.  If the tracking get's fast, it's because the device itself alters it's advertising based on the connection to the app.

yuri-vashchenko, 2022-03-12:
> Can you explain how you got it working? I'm trying latest release but it's the same, only without qryerr in the log

I did nothing special, just installed newer binaries and it is working fine now. See my configuration below:

[image]

DeekB1983, 2022-03-12:
> Can you explain how you got it working? I'm trying latest release but it's the same, only without qryerr in the log

What app are you using for you MiBand, Are you using MiFit?

I had an issue originally where the mi band was not detected and I found I had to go into the settings in the mifit app and make the mi band discoverable before it would be detected.

ale1800, 2022-03-13:
> > > > As far as I know the only devices so far that support querying is apple devices. I wouldn't waste airtime trying to query mifits especially since they use fixed mac addresses as far as I know. Can this be closed or are you still having issues on the latest firmware?
> > > 
> > > 
> > > Yes, it works fine on latest release
> > 
> > Can you explain how you got it working? I'm trying latest release but it's the same, only without qryerr in the log
> 
> What app are you using for you MiBand, Are you using MiFit?
> 
> I had an issue originally where the mi band was not detected and I found I had to go into the settings in the mifit app and make the mi band discoverable before it would be detected.
> 
> To make it discoverable from  Mi Fit app.
> Open the app
> choose profile 
> press on your mi device so it takes you to setting
> Scroll down and you should have an option for discoverable
> Change to on 

Thanks for replying. Already have it discoverable but still unusable. In the logs is quite fast to recognize but if I'm close to the esp It randomly switches between left and close, so it's not reliable. Then I found that mqtt updates are really really slow, even if I set to send updates only for my id.

closed 2022-05-19 by stale[bot]