ESPresense/ESPresense #271: Repeated crashing (core 1 panic and corrupt heap)

dodgeboy, 2022-02-14:
On v2.0.23, all of my ESP-WROOM-32s are crashing every few minutes with the following in the logs:

Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (BLE Scan) 
Core 1 register dump:
PC      : 0x4012ae37  PS      : 0x00060936  A0      : 0x80124979  A1      : 0x3ffdcce0  
A2      : 0x3ffdc770  A3      : 0x3ffdd020  A4      : 0x3ffdcf04  A5      : 0x3f40bd0c  
A6      : 0x00000001  A7      : 0x3ffdcf44  A8      : 0x462d4100  A9      : 0x3ffdcfd0  
A10     : 0x00000003  A11     : 0x00060923  A12     : 0x00060920  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x4012ae37:0x3ffdcce0 0x40124976:0x3ffdcd10 0x401204dd:0x3ffdd020 0x400edc25:0x3ffdd0e0 0x400e6504:0x3ffdd130 0x400f12eb:0x3ffdd190 0x400f1355:0x3ffdd1c0 0x400f13b6:0x3ffdd200 0x400f13b6:0x3ffdd240 0x400f13b6:0x3ffdd280 0x400f13b6:0x3ffdd2c0 0x400f13b6:0x3ffdd300 0x400f13b6:0x3ffdd340 0x400f13b6:0x3ffdd380 0x400f13b6:0x3ffdd3c0 0x400f13b6:0x3ffdd400 0x400f13b6:0x3ffdd440 0x400f13b6:0x3ffdd480 0x400f13b6:0x3ffdd4c0 0x400f13b6:0x3ffdd500 0x400f13b6:0x3ffdd540 0x400f13b6:0x3ffdd580 0x400f13b6:0x3ffdd5c0 0x400f13b6:0x3ffdd600 0x400f13b6:0x3ffdd640 0x400f13b6:0x3ffdd680 0x400f13b6:0x3ffdd6c0 0x400f13b6:0x3ffdd700 0x400f13b6:0x3ffdd740 0x400f13b6:0x3ffdd780 0x400f13b6:0x3ffdd7c0 0x400f13b6:0x3ffdd800 0x400f13b6:0x3ffdd840 0x400f13b6:0x3ffdd880 0x400f13b6:0x3ffdd8c0 0x400f13b6:0x3ffdd900 0x400f13b6:0x3ffdd940 0x400f13b6:0x3ffdd980 0x400f13b6:0x3ffdd9c0 0x400f13b6:0x3ffdda00 0x400f13b6:0x3ffdda40 0x400f13b6:0x3ffdda80 0x400f13b6:0x3ffddac0 0x400f13b6:0x3ffddb00 0x400f13b6:0x3ffddb40 0x400f13b6:0x3ffddb80 0x400f13b6:0x3ffddbc0 0x400f13b6:0x3ffddc00 0x400f13b6:0x3ffddc40 0x400f13b6:0x3ffddc80 0x400f13b6:0x3ffddcc0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828

And occasionally this message:
CORRUPT HEAP: Bad head at 0x3ffe3d50. Expected 0xabba1234 got 0x3ffe3e90
abort() was called at PC 0x40086cdd on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f658:0x3ffdd3e0 0x4008f8d5:0x3ffdd400 0x40086cdd:0x3ffdd420 0x40086e09:0x3ffdd450 0x40126813:0x3ffdd470 0x401202b9:0x3ffdd730 0x40120225:0x3ffdd780 0x400943ed:0x3ffdd7b0 0x40081e6e:0x3ffdd7d0 0x40086bd5:0x3ffdd7f0 0x4000bec7:0x3ffdd810 0x401a9fdd:0x3ffdd830 0x400e05b9:0x3ffdd850 0x400e070e:0x3ffdd870 0x400f1147:0x3ffdd8a0 0x400f5e21:0x3ffdd900 0x400d72da:0x3ffdda50 0x400908e6:0x3ffddad0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

wazoo, 2022-02-14:
I am also running into this bug now on two different ESP-WROOM-32 devices that were working previously.  I have one still attached to a computer and it is boot looping for a while as described above and I have 5 others that have locked up from what I can tell (they are all disconnected from home-assistant now) and restarting them seems to keep them connected for a few min.

I am pretty sure that these updated from v2.0.22 or v2.0.21 overnight like the reporter said and I saw something similar on the previous versions _when query devices was enabled_ which I can't seem to disable now on this new version but it was disabled before the upgrade (I never had a chance to open an issue about that but maybe it is related).

Please let me know what logs I can add that would be helpful, my traces look very similar to the above except different memory addresses.

casaerreti, 2022-02-14:
Hi, just to confirm same here. Fresh install on 3 boards with .23 version. All the same. Sending positive vibes

droptopgokart, 2022-02-14:
Is this what caused all my ESPresense ESP's to drop offline this morning. Not even showing up on my network.  I Figure it has to be a wide spread issue to make them all drop at one time.

briancmoses, 2022-02-14:
I'm seeing the same on my ESP32 ESP-WROOM-32 that has been reported by @dodgeboy and @wazoo.   Brand new fresh installs on new boards.

Please let us know if there's any additional information we can gather!

mgzob, 2022-02-14:
At risk of being another "me too" comment - I just found this after installing ESPresence for the first time today. Same manifestation:

Also a WROOM-32; 

`lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)` appears periodically. 

Crash follows format: 

> [11:00:19]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> [11:00:19]Core 0 register dump:
> [11:00:19]PC      : 0x400efb3c  PS      : 0x00060930  A0      : 0x800efd8f  A1      : 0x3ffe33c0  
> [11:00:19]A2      : 0x00000000  A3      : 0x00000016  A4      : 0x00000000  A5      : 0x00000000  
> [11:00:19]A6      : 0x3ffe354c  A7      : 0xff000000  A8      : 0x801a9fe0  A9      : 0xad567800  
> [11:00:19]A10     : 0x000000ba  A11     : 0x00000001  A12     : 0x00000014  A13     : 0x3f403f90  
> [11:00:19]A14     : 0xad567800  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
> [11:00:19]EXCVADDR: 0xad567800  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

DTTerastar, 2022-02-14:
2.0.24 fixes this.  Sorry for the trouble.

gerardwr, 2022-02-14:
Installed for the 1st time yesterday, worked fine, excellent firmware.

Had this rebooting issue since 1 day after the Esp32 firmware was automagically updated.

Installed 2.0.24 but it still reboots after approximately 1 minute.

Not sure if it is related to this issue

I get the following error in the log every time  :-(

In the log:
```20:18:28.385 -> Verbose | apple:1006:10-12                                          MD: 4c001006131e5dd8fccd
20:18:28.385 -> I (41441) wifi:flush txq
20:18:28.385 -> I (41442) wifi:stop sw txq
20:18:28.385 -> I (41443) wifi:lmac stop hw txq
20:18:28.385 -> ets Jun  8 2016 00:22:57
20:18:28.385 -> 
20:18:28.385 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:18:28.418 -> configsip: 0, SPIWP:0xee
20:18:28.418 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:18:28.418 -> mode:DIO, clock div:2
20:18:28.418 -> load:0x3fff0018,len:4
20:18:28.418 -> load:0x3fff001c,len:1044
20:18:28.418 -> load:0x40078000,len:8896
20:18:28.418 -> load:0x40080400,len:5828
20:18:28.418 -> entry 0x400806ac
20:18:29.061 -> Connecting to WiFi (30:AE:A4:5D:B3:F0)...
```

briancmoses, 2022-02-14:
> 2.0.24 fixes this. Sorry for the trouble.

Indeed it does,  it resolved my issues.  Thanks for resolving this issue,  I appreciate it!

dotr, 2022-02-15:
I'm running v2.0.25 and it's crashing constantly after fresh install. I'm running Nodemcu ESP-32S ESP-WROOM-32. Managed to get my Wifi and MQTT settings in, it was crashing and rebooting before, but one time it worked long enough so I hurried to fill all the configs.
Now it crashes when connecting to Wifi or after several minutes if it manages to connect.

Log:
```
Connecting to WiFi SSID '*********'........................................................................................Guru Meditation Error: Core  panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC       0x4008da  PS       0x00060634  A0       0x80139014  A1       0x3fdf0  
A2       0x3fd3d  A3       0x00000000  A4       0x00000000  A5       0x00000020  
A6       0x3fcb8  A7       0x3fb76c A8       0x8018bc  A9       0x3fd000  
A10      0x00000001  A11      0x00000000  A12      0x3fddc A13      0x00000000  
A14      0x00000000  A15      0x00000000  SAR      0x00000020  EXCCAUSE 0x00000005  
EXCVADDR 0x00000000  LBEG     0x4000ce  LEND     0x4000cf  LCOUNT   0xffff  

ELF file SHA256: 000000000000000

Backtrace:0x4008da:0x3fdf00x40139011:0x3fd070 0x401335b:0x3fd090 0x401344c:0x3fd0c 0x40134d5:0x3fd100 0x40134e1:0x3fd130 0x401396c:0x3fd160 0x400908e:0x3fd190

ore 1 register dump:
PC       0x401d852  PS       0x00060234  A0       0x8018c16  A1       0x3fbb0  
A2       0x00000000  A3       0x80000001  A4       0x00000000  A5       0x00000001  
A6       0x00060a0  A7       0x00000000  A8       0x00060023  A9       0x3fb76c 
A10      0x00000000  A11      0x00000001  A12      0x80090a8  A13      0x3fc370  
A14      0x00000000  A15      0x3fb8e  SAR      0x00000000  EXCCAUSE 0x00000005  
EXCVADDR 0x00000000  LBEG     0x00000000  LEND     0x00000000  LCOUNT   0x00000000  

ELF file SHA256: 000000000000000

Backtrace:0x401d852:0x3fbb00x4018c13:0x3fbc0 0x40092095:0x3fbc0 0x400908e:0x3fbc0

Rebooting...

```

freman, 2022-02-15:
I'm going to add a voice to the v2.0.25 problem, I haven't gone and hooked any of them up to a serial terminal yet but all 3 of mine are offline

DTTerastar, 2022-02-15:
We're making some progress: https://github.com/h2zero/NimBLE-Arduino/issues/353

ChirpyTurnip, 2022-02-15:
Running on v2.0.26 and I'm no longer crashing and locking, but after 'running' overnight from 7pm to 7am the uptime is only reported as being low - so it is still rebooting:

IP - 192.168.20.205
Uptime - 5,044
Firm - esp32
Rssi - -34
Ver - v2.0.26
Adverts - 16,941
Seen - 16,864
Queried - 612
Reported - 7,772
FreeHeap - 91,336
MinFreeHeap - 43,728
MaxAllocHeap - 38,264
ResetReason - SoftwareCpu

So progress...but not yet stable?

Settings:

☑Status LED
☑Automatically Update
      Arduino OTA Update
☑Home Assistant Discovery
☑Active scanning
☑Send to telemetry topic
☑Send to rooms topic
☑Send to devices topic

Just in case it helps.....

CP.

droptopgokart, 2022-02-19:
Did we ever come up with a solution for this issue? All of mine are still dropping offline.

ChirpyTurnip, 2022-02-19:
I'm running v2.0.28 and it seems pretty stable....I had a 1 second drop out in the last 24 hours and that's too short to be crash/reboot....most likely just a DHCP renewal forcing an IP change. 

I did see however that support for temperature and humidity sensors etc has vanished...there is only motion sensor support now....I don't know if that is coming back or if that was part of the problem...

I know that's a crappy "well it works for me" answer.....but I can confirm that it was crashing and rebooting like crazy before and now it is a lot happier. I'm running it on a ESP32-WROOM-32U DevKit C (from mouser, so a 100% genuine one).

DTTerastar, 2022-03-22:
Everybody should use 2.1, it's much improved.

closed 2022-03-22 by DTTerastar