ESPresense/ESPresense #272: M5Stack ATOM Lites Becoming Unavailable

michaelfucci, 2022-02-14:
All 9 of my M5Stack Atom Lites are becoming unavailable within a short period of time after boot. Running ESPresence v2.0.23. Example config attached.

[image]

daveyijzermans, 2022-02-14:
Having this problem as well. Reverting to an older version (I picked v2.0.15) and disabling auto update until this is fixed.

Zipties, 2022-02-14:
The auto update did this to all my ESPs. Reloading the firmware via USB "seems" to have resolved it. However even though I have query devices se tot zero (also tried blank) it seems to still be querying everything it finds. 

''29 WiFi networks found.
[E][WebServer.cpp:633] _handleRequest(): request handler not found
29 WiFi networks found.
ets Jun  1 Query | MAC: beffe701084d, ID: beffe701084d                                                 rssi -80
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
1 QryErr| MAC: beffe701084d, ID: beffe701084d                                                 rssi -80, try 4, retry after 19000ms
1 Query | MAC: 365d74e5c6aa, ID: apple:0906:10                                                rssi -43
1 QryErr| MAC: 365d74e5c6aa, ID: apple:0906:10                                                rssi -43, try 2, retry after 1000ms
0 New Q | MAC: 6688dce29664, ID: apple:1006:10-1                                              061d8a95ef18
0 New Q | MAC: f8042ee37d9f, ID: samsung:f8042ee37d9f                                         
1 Query | MAC: e0c24e10c365, ID: md:0059:8                                                    rssi -68
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
1 QryErr| MAC: e0c24e10c365, ID: md:0059:8                                                    rssi -68, try 1, retry after 5000ms
1 Query | MAC: 64ad81abd094, ID: sd:0xfe9f                                                    rssi -90
1 QryErr| MAC: 64ad81abd094, ID: sd:0xfe9f                                                    rssi -90, try 1, retry after 5000ms
0 New Q | MAC: 84ee0348dd17, ID: iBeacon:74278bda-b644-4520-8f0c-720eaf059935-0-16411         
0 New Q | MAC: 6b55ccd7d15b, ID: md:ffff:1815                                                 
1 Query | MAC: 670a0bd99671, ID: apple:0719:29                                                rssi -50''

Zipties, 2022-02-14:
Never mind, they're all still going offline. 

They seem to be panic rebooting before going completely offline 

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (4C:75:25:C6:8E:94)...
I (72) wifi:wifi driver task: 3ffd43c0, prio:23, stack:3584, core=0
I (369) wifi:wifi firmware version: dc30037
I (370) wifi:config NVS flash: enabled
I (370) wifi:config nano formating: disabled
I (370) wifi:Init data frame dynamic rx buffer num: 32
I (375) wifi:Init management frame dynamic rx buffer num: 32
I (380) wifi:Init management short buffer num: 32
I (385) wifi:Init dynamic tx buffer num: 32
I (389) wifi:Init static rx buffer size: 1600
I (393) wifi:Init static rx buffer num: 16
I (396) wifi:Init dynamic rx buffer num: 32
I (490) wifi:mode : sta (4c:75:25:c6:8e:94)
Connecting to WiFi SSID 'BlackMesa'..............I (2560) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2561) wifi:state: init -> auth (b0)
I (2570) wifi:state: auth -> assoc (0)
I (2579) wifi:state: assoc -> run (10)
.I (2623) wifi:connected with BlackMesa, aid = 16, channel 11, BW20, bssid = 36:9f:07:00:1b:8f
I (2623) wifi:security type: 3, phy: bgn, rssi: -36
I (2633) wifi:pm start, type: 1

I (2643) wifi:AP's beacon interval = 102400 us, DTIM period = 2
192.168.40.3
Version:     v2.0.23
IP address:   192.168.40.3
DNS address:  192.168.40.91
Hostname:     espresense-living-room
Room:         Living Room
Max Distance: 2.00
Telemetry:    enabled
Rooms:        enabled
Devices:      enabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        
Include:      exp:20 apple:1005:9-26 apple:iphone13-2 apple:1005:9-24
Exclude:      
Connecting to MQTT 192.168.40.195 1883
Verbose | 6dc4be6bb5b9                                              MD: 4c000c0e00353a18d2aefb6639c488529fb71006711d0008d8c8
Verbose | beffe701084d                                              AD: 0x1812
Verbose | 56a4a4513e3f                                              AD: 0xfe9f
Verbose | ad:0xfe9f                                                 SD: 0xfe9f/0000000000000000000000000000000000000000
Verbose | 4529d8a89f22                                              MD: 4c0010052598ebb7f1
Verbose | 4e0a540773be                                              MD: 060001092002449181a26ecbceb51e1347d8df0bc03a86dd2603484e80
Verbose | beffe701084d                                              AD: 0x1812
[W][AsyncTCP.cpp:930] _poll(): pcb is NULL
Verbose | 52fcb58905a0                                              MD: 4c001006061d27a41828
Verbose | 6c4a8528e329                                              MD: 4c00100501140e7c11
Verbose | beffe701084d                                              AD: 0x1812
0 New Q | MAC: beffe701084d, ID: beffe701084d                                                 
Verbose | 00126f589c1a                                              AD: 0xfe8f
Verbose | 00126f589c1a                                              MD: 1000
[... 190 lines cut ...]
Verbose | beffe701084d                                              AD: 0x1812
Verbose | 00126f589c1a                                              AD: 0xfe8f
Verbose | 00126f589c1a                                              MD: 1000
Verbose | apple:1007:11-7                                           MD: 4c001007391f7260c0b458
Verbose | beffe701084d                                              AD: 0x1812
Verbose | sd:0xfe9f                                                 AD: 0xfe9f
Verbose | sd:0xfe9f                                                 SD: 0xfe9f/0000000000000000000000000000000000000000
1 Query | MAC: 755b0c6dfec8, ID: sd:0xfe9f                                                    rssi -54
Verbose | sd:0xfe9f                                                 AD: 0xfe9f
Verbose | sd:0xfe9f                                                 SD: 0xfe9f/0000000000000000000000000000000000000000
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400efea0  PS      : 0x00060130  A0      : 0x800f00f3  A1      : 0x3ffe2b50  
A2      : 0x00000000  A3      : 0x00000016  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffe2cd8  A7      : 0xff000000  A8      : 0x801aac68  A9      : 0xad567800  
A10     : 0x000000ba  A11     : 0x00000001  A12     : 0x00000014  A13     : 0x3f404864  
A14     : 0xad567800  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xad567800  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400efea0:0x3ffe2b50 0x400f00f0:0x3ffe2b70 0x400f4da6:0x3ffe2b90 0x400f6662:0x3ffe2df0 0x400d24ac:0x3ffe2e10 0x400e0c31:0x3ffe2e30 0x400e4192:0x3ffe2e70 0x400e906d:0x3ffe2ee0 0x400e9119:0x3ffe2f20 0x400e9365:0x3ffe2f40 0x400e7c1d:0x3ffe2f60 0x400eed8e:0x3ffe2f80 0x400e0157:0x3ffe2fa0 0x400908e6:0x3ffe2fc0

Rebooting...
```

hellcry37, 2022-02-14:
same issue using esp32 dev 1

Zipties, 2022-02-14:
here is a hard crash that it does not recover from 

```
Verbose | beffe701084d                                              AD: 0x1812
Verbose | c2356b36c725                                              MD: 4c0012020002
Verbose | msft:cdp:0902                                             MD: 060001092002cd2210b7966c7c8008a491a44efad71dbf6bd72a9c2e00
Verbose | beffe701084d                                              AD: 0x1812
Verbose | exp:20                                                    AD: 0xfd6f
Verbose | exp:20                                                    SD: 0xfd6f/ded09ae03646dde062361330a3b321a7ed14668e
Verbose | sd:0xfe9f                                                 AD: 0xfe9f
Verbose | sd:0xfe9f                                                 SD: 0xfe9f/0000000000000000000000000000000000000000
Verbose | sd:0xfe9f                                                 AD: 0xfe9f
Verbose | sd:0xfe9f                                                 SD: 0xfe9f/0000000000000000000000000000000000000000
Verbose | beffe701084d                                              AD: 0x1812
Verbose | apple:1005:9-26                                           MD: 4c00100521380abf39
Verbose | apple:1007:11-7                                           MD: 4c001007301f04e72fd568
Verbose | beffe701084d                                              AD: 0x1812
Verbose | md:0059:8                                                 MD: 5900e0c24e10c365
1 Query | MAC: e0c24e10c365, ID: md:0059:8                                                    rssi -74
***ERROR*** A stack overflow in task BLE Scan has been detected.
abort() was called at PC 0x4008f8ec on core 1
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr 
Guru Meditation Error: Core  1 panic'ed (Double exception)
```

DTTerastar, 2022-02-14:
2.0.24 fixes this.  It was querying every device and there are some bugs in NimBLE that cause horrible crashes querying certain devices that don't expect to be queried.

michaelfucci, 2022-02-14:
> 2.0.24 fixes this. It was querying every device and there are some bugs in NimBLE that cause horrible crashes querying certain devices that don't expect to be queried.

After several reboots, some of my devices updated to v2.0.25. Unfortunately, they are still having the same issue.

DTTerastar, 2022-02-15:
We're making some progress: https://github.com/h2zero/NimBLE-Arduino/issues/353

closed 2022-03-22 by DTTerastar