ESPresense/ESPresense #278: Reboot loop is back with v2.0.26

dewoert, 2022-02-15:
**Describe the bug**

Since the update to v2.0.26 the reboot-loop is back.
Was working fine with v2.0.25, but all devices have been updated now to v2.0.26, and all reboot after about 30sec.

**Config**

- Version: 2.0.26
- Active scan enabled: yes
- Include filter: n/a
- Exclude filter: n/a
- Query filter: apple:1007:11-7 apple:1006:10-12

**Hardware**

- Firmware flavor: ESP32
- Device (be specific): ESP32-D0WDQ6

**Logs**

```
0 New Q | MAC: 667fa3d5fcf3, ID: apple:1007:11-7                                              7a1fdf89a53148
0 New   | MAC: 4da0db5aed20, ID: apple:1005:9-26                                              2a98c98191
E (11348) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11348) task_wdt:  - IDLE0 (CPU 0)
E (11348) task_wdt: Tasks currently running:
E (11348) task_wdt: CPU 0: btController
E (11348) task_wdt: CPU 1: IDLE1
E (11348) task_wdt: Aborting.
abort() was called at PC 0x4018b210 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f658:0x3ffbfee0 0x4008f8d5:0x3ffbff00 0x4018b210:0x3ffbff20 0x4008dd01:0x3ffbff40 0x4000bfed:0x3ffb5e20 0x40091a59:0x3ffb5e30 0x40084223:0x3ffb5e50 0x4018768d:0x3ffb5e70 0x40187770:0x3ffb5e90 0x4018c57d:0x3ffb5ec0 0x4018c63d:0x3ffb5ef0 0x4013ebab:0x3ffb5f20 0x40191df2:0x3ffb5f40 0x400908e6:0x3ffb5f70

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

```

theidol, 2022-02-15:
Not Good, reboot loop

tslpre, 2022-02-15:
I can confirm the same behaviour on all the ESP32 that are now on 2.0.26. A few MACs are identified and then I can see the "Task watchdog got triggered" issue as per logs above.

RutgerDiehard, 2022-02-15:
I have noticed a reboot loop on all 8 of my Atom Lites after update to 2.0.26. They won't stay up for longer than 30 seconds or before I see them reconnect to my Wi-Fi network. They were fine on 2.0.25 and previous versions.

leacho73, 2022-02-15:
ESP32 updated to latest 2.0.26 overnight and is now in a constant reboot loop

dewgenenny, 2022-02-15:
Same here, ESP32-WROOM-32D, 2.0.26, active scanning, no filters.

`E (8039) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (8039) task_wdt:  - IDLE0 (CPU 0)
E (8039) task_wdt: Tasks currently running:
E (8039) task_wdt: CPU 0: BLE Scan
E (8039) task_wdt: CPU 1: IDLE1
E (8039) task_wdt: Aborting.
abort() was called at PC 0x4018b210 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f658:0x3ffbfee0 0x4008f8d5:0x3ffbff00 0x4018b210:0x3ffbff20 0x4008dd01:0x3ffbff40 0x4000bfed:0x3ffde2b0 0x40091a59:0x3ffde2c0 0x40093c4f:0x3ffde2e0 0x400943b6:0x3ffde300 0x40081dd1:0x3ffde320 0x40081e02:0x3ffde340 0x40086bc9:0x3ffde360 0x4000beaf:0x3ffde380 0x401a9e33:0x3ffde3a0 0x400f6d53:0x3ffde3c0 0x400d73c1:0x3ffde3e0 0x400908e6:0x3ffde460
`

Rebooting every 20 seconds or less.

JameZUK, 2022-02-15:
Same, none of the 6 I have are working now...

nijez, 2022-02-15:
我刚升级就发现ESP32一直在重启，没办法只好退回2.0.25版本

nijez, 2022-02-15:
现在都把自动更新关闭了

shamimrahim, 2022-02-15:
Same as above with all 5 of my ESP32s

gooselk, 2022-02-15:
I didn't check the terminal log yet but it would explain that all my ESP32 began spamming my broker with new connections this morning all at once.

I'll check this when I get home.

JameZUK, 2022-02-15:
I think automatic updates will be disabled on my set from now on too. Annoyingly several have dropped off the network completely so its a bit of a manual job to restore them. Still a great peice of work though, its been solid for a long time for me. :)

viab23, 2022-02-15:
Same here, but I am not in a reboot loop but it switches to AP mode after the first crash, as it cannot reconnect to the wifi somehow?!

`E (8815) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (8815) task_wdt:  - IDLE0 (CPU 0)
E (8815) task_wdt: Tasks currently running:
E (8815) task_wdt: CPU 0: btController
E (8815) task_wdt: CPU 1: IDLE1
E (8815) task_wdt: Aborting.
abort() was called at PC 0x4018b210 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f658:0x3ffbfee0 0x4008f8d5:0x3ffbff00 0x4018b210:0x3ffbff20 0x4008dd01:0x3ffbff40 0x4000bfed:0x3ffb5eb0 0x40091a59:0x3ffb5ec0 0x40081c7b:0x3ffb5ee0 0x4008d95a:0x3ffb5f00 0x40191839:0x3ffb5f20 0x40191e29:0x3ffb5f40 0x400908e6:0x3ffb5f70

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (30:AE:A4:06:F5:28)...
Connecting to WiFi SSID 'MyWifi'................................................................................................................................................................................................................................................................................................................................................................................................................ failed.
Starting access point for configuration portal.
SSID: 'espresense-office'
IP: 192.168.4.1`

smithdo, 2022-02-15:
E (8978) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (8978) task_wdt:  - IDLE0 (CPU 0)
E (8978) task_wdt: Tasks currently running:
E (8978) task_wdt: CPU 0: BLE Scan
E (8978) task_wdt: CPU 1: IDLE1
E (8978) task_wdt: Aborting.
abort() was called at PC 0x4018b210 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f658:0x3ffbfee0 0x4008f8d5:0x3ffbff00 0x4018b210:0x3ffbff20 0x4008dd01:0x3ffbff40 0x4000bfed:0x3ffde220 0x40091a59:0x3ffde230 0x40093c4f:0x3ffde250 0x400943b6:0x3ffde270 0x40081dd1:0x3ffde290 0x40081e02:0x3ffde2b0 0x40086bc9:0x3ffde2d0 0x4000beaf:0x3ffde2f0 0x401a9e33:0x3ffde310 0x400f6d53:0x3ffde330 0x400d73c1:0x3ffde350 0x400908e6:0x3ffde3d0

Rebooting...

mario-almeida, 2022-02-15:
Same issue
E (8921) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (8921) task_wdt:  - IDLE0 (CPU 0)
E (8921) task_wdt: Tasks currently running:
E (8921) task_wdt: CPU 0: BLE Scan
E (8921) task_wdt: CPU 1: IDLE1
E (8921) task_wdt: Aborting.
abort() was called at PC 0x4018b210 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f658:0x3ffbfee0 0x4008f8d5:0x3ffbff00 0x4018b210:0x3ffbff20 0x4008dd01:0x3ffbff40 0x400d66ad:0x3ffddf20 0x400d73d1:0x3ffde420 0x400908e6:0x3ffde4a0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

carlywarly, 2022-02-15:
Same, none of my 5 I have are working now...

jimmyeao, 2022-02-15:
Not sure if this is relevant, but 2 of my devices seem to be exhibiting this behaviour, 2 seem ok. All the same board and version
Only difference is wifi strength
[image]

GoNzCiD, 2022-02-15:
Same here:
```
E (9084) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (9084) task_wdt:  - IDLE0 (CPU 0)
E (9084) task_wdt: Tasks currently running:
E (9084) task_wdt: CPU 0: btController
E (9084) task_wdt: CPU 1: IDLE1
E (9084) task_wdt: Aborting.
abort() was called at PC 0x4018be40 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f658:0x3ffbff60 0x4008f8d5:0x3ffbff80 0x4018be40:0x3ffbffa0 0x4008dd01:0x3ffbffc0 0x4000bfed:0x3ffdf9f0 0x40091a59:0x3ffdfa00 0x40084223:0x3ffdfa20 0x401882bd:0x3ffdfa40 0x401883a0:0x3ffdfa60 0x4018d1ad:0x3ffdfa90 0x4018d26d:0x3ffdfac0 0x4013f7db:0x3ffdfaf0 0x40192a22:0x3ffdfb10 0x400908e6:0x3ffdfb40

Rebooting...
ets Jun  8 2016 00:22:57
```

mario-almeida, 2022-02-15:
Reverted to v2.0.25 and disabled auto update, now stable.

dewoert, 2022-02-16:
Reboot loop seems resolved with v2.0.27
Other problem now : https://github.com/ESPresense/ESPresense/issues/289

closed 2022-02-16 by dewoert