ESPresense/ESPresense #279: Apple Watch Series 6 - Query Error

leacho73, 2022-02-15:
**Describe the bug**

I have noticed whilst using ESPresense on an ESP32, whilst I get inconsistent results with my iPhone (known issue, being worked on) - I am also seeing in the logs that I just get a 'QueryErr' when trying to poll my Apple Watch Series 6. I was hoping to use the watch as a more reliable method of presence detection as you would normally be wearing that, rather than having to guarantee having a phone on you all the time.

**Config**

- Version: 2.0.26

**Hardware**

ESP32 NodeMCU

**Logs**

0 QryErr| MAC: ###############, ID: apple:1005:9-26 rssi -56, try 1, retry after 0ms
0 QryErr| MAC: ###############, ID: apple:1005:9-26  rssi -52, try 1, retry after 0ms
0 New Q | MAC: ###############, ID: apple:1005:9-26 21984fa935
0 Query | MAC: ###############, ID: apple:1005:9-26  rssi -51
0 QryErr| MAC: ###############, ID: apple:1005:9-26 rssi -51, try 2, retry after 1000ms
0 Query | MAC: ###############, ID: apple:1005:9-26 rssi -60
0 QryErr| MAC: ###############, ID: apple:1005:9-26 rssi -60, try 3, retry after 5000ms

**Screenshots**

Zipties, 2022-02-16:
Query is basically broken right now

DTTerastar, 2022-03-10:
I have gotten apple watches to work a bit better, but still trying to figure this out.  I've gotten a nice packet dump of it happening: https://github.com/h2zero/NimBLE-Arduino/issues/366

DTTerastar, 2022-06-27:
Anyone but me have a chance to test this?

DTTerastar, 2022-07-07:
The newest release has this in it.  I think we should probably close it. @leacho73 ? @Zipties ?

leacho73, 2022-07-11:
Happy for it to be closed.

closed 2022-07-12 by DTTerastar