ESPresense/ESPresense #281: Disable red LED on ESP-WROOM-32?

dodgeboy, 2022-02-15:
It would be nice if there was an option to disable the red power light on the ESP-WROOM-32, similar to how the activity/status LED can be disabled.  Even when in a 3D printed case, that LED lights up the whole case and draws attention to the unit.  I'm not sure if this is something that is possible within the firmware, but if it is, I would really like to see the feature added.

dewgenenny, 2022-02-15:
I have a feeling that the red power light is not software steerable, it's just on, because power is on... 
 see reddit thread here about it too: https://www.reddit.com/r/esp32/comments/7kyfpx/turn_off_led/

Maybe you can try removing it? I have successfully scraped those tiny LEDs off boards before with no impact...

dodgeboy, 2022-02-15:
> I have a feeling that the red power light is not software steerable, it's just on, because power is on... see reddit thread here about it too: [...]

Thanks for the reply.  I was afraid of that.  It should be easy enough to physically disable, though.

closed 2022-02-15 by dodgeboy