ESPresense/ESPresense #283: Multiple reboots per minute on v2.0.26 -- all units

cbhiii, 2022-02-15:
**Describe the bug**

Multiple reboots per minute on v2.0.26.

**Config**

- Version: v2.0.26
- Active scan enabled: No
- Include filter: apple:1005:9-24 apple:1005:9-26
- Exclude filter:
- Query filter: apple:1005:9-24 apple:1005:9-26

**Hardware**

- Firmware flavor: Flashed from website on Sunday 2/13.
- Device (be specific): ESP-WROOM-32

**Logs**

```
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Connecting to WiFi (78:E3:6D:10:64:B0)...
Connecting to WiFi SSID '3264128'...............192.168.32.168
Version:     v2.0.26
IP address:   192.168.32.168
DNS address:  192.168.32.3
Hostname:     espresense-office
Room:         office
Max Distance: 2.50
Telemetry:    disabled
Rooms:        enabled
Devices:      disabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        apple:1005:9-24 apple:1005:9-26
Include:      apple:1005:9-24 apple:1005:9-26
Exclude:      
Connecting to MQTT 192.168.32.6 1883
0 New   | MAC: 0fc6a1315918, ID: msft:cdp:0902                                                b2c1b822c7e0de6ba64b0648432f59766973fe64
0 New   | MAC: 43e22410f0b0, ID: apple:1002:6-12                                              0300
0 New   | MAC: 64ff0a0edd8c, ID: 64ff0a0edd8c                                                 
0 New   | MAC: 7e2bc895b8e5, ID: apple:1005:9-12                                              051cb02f07
0 New   | MAC: 2e64a49772d8, ID: exp:20                                                       a43e488126d61cbe9ad08b48548458270d6a83cc
0 New   | MAC: 68d79a3f4440, ID: 68d79a3f4440                                                 
0 New Q | MAC: 7781fbc641dc, ID: apple:1005:9-24                                              0098e07c2d
E (8001) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (8001) task_wdt:  - IDLE0 (CPU 0)
E (8001) task_wdt: Tasks currently running:
E (8001) task_wdt: CPU 0: btController
E (8001) task_wdt: CPU 1: IDLE1
E (8001) task_wdt: Aborting.
abort() was called at PC 0x4018b210 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f658:0x3ffbfee0 0x4008f8d5:0x3ffbff00 0x4018b210:0x3ffbff20 0x4008dd01:0x3ffbff40 0x4000bfed:0x3ffb5e20 0x40091a59:0x3ffb5e30 0x40084223:0x3ffb5e50 0x4018768d:0x3ffb5e70 0x40187770:0x3ffb5e90 0x4018c57d:0x3ffb5ec0 0x4018c63d:0x3ffb5ef0 0x4013ebab:0x3ffb5f20 0x40191df2:0x3ffb5f40 0x400908e6:0x3ffb5f70

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
[... 878 lines cut ...]
Connecting to WiFi (78:E3:6D:10:64:B0)...
Connecting to WiFi SSID '3264128'...............192.168.32.168
Version:     v2.0.26
IP address:   192.168.32.168
DNS address:  192.168.32.3
Hostname:     espresense-office
Room:         office
Max Distance: 2.50
Telemetry:    disabled
Rooms:        enabled
Devices:      disabled
Discovery:    enabled
PIR Sensor:   disabled
Radar Sensor: disabled
Query:        apple:1005:9-24 apple:1005:9-26
Include:      apple:1005:9-24 apple:1005:9-26
Exclude:      
Connecting to MQTT 192.168.32.6 1883
0 New   | MAC: 3bc153c520d4, ID: msft:cdp:0902                                                9b1067141f195aad6b4d8b4e705f0a00d63d0ec4
0 New   | MAC: 68d79a3f4440, ID: 68d79a3f4440                                                 
0 New   | MAC: 7145bbac00e6, ID: apple:1002:6-12                                              0300
0 New Q | MAC: 7781fbc641dc, ID: apple:1005:9-24                                              0098e07c2d
0 Query | MAC: 7781fbc641dc, ID: apple:1005:9-24                                              rssi -62

```

ahmedsamer, 2022-02-16:
Hello, 

I'm still using v2.0.19 because of this bug. It seems to be fixed in v2.0.27 [#284](https://github.com/ESPresense/ESPresense/pull/284) but introduced another bug #289

DTTerastar, 2022-03-22:
Everybody should use 2.1, it's much improved.

closed 2022-03-22 by DTTerastar